DroneBridge/ESP32 #1: UAV couldn't be found on the Android App

dardrai, 2019-01-21:
I had my fair share of problems to get the ESP32 to run with the DroneBridge Software (had to download the Master and flash with the esp-idf) but now it works (or I believe it does). But atm I can't connect to the drone with your app - the UAV wasn't found or didn't it responde in time? When I connect to the ESP32 with my Browser i see the fallowing (and changed some smaller details like AP Password): https://imgur.com/KWYixoI on Inav I set my ports like this: https://imgur.com/dtpr6Ii (UART4 is where the ESP32 is connected) and in the config Tab i set it like this: https://imgur.com/a/hEfb25D (GPS and Compass are deactivated but will added soon) - I did connect the ESP to my Kakute F7 like this: https://imgur.com/TNrPv0I (5V to 5V, Grd to Grd, TX (ESP32) to UART4 RX, RX (ESP32) to UART4 TX) I even tried to connect TX to TX and RX to RX (maybee you'll see it on this foto: https://imgur.com/ikUdyc1) but either way I couldn't connect with the app :( Full view of the ESP32 - it is an ESP-WROOM-32: https://imgur.com/YAJ54Rh) Here are also the screenshots of the App and Settings: https://imgur.com/pbO8sHo , https://imgur.com/aGsMFn9 , https://imgur.com/li5auFG --- Before some one ask why there is an ESP8266 on uart1 => this is atm the only why to connect to my flight controller after the USB Port was ripped off :( shout out to inav to leave UART1 with MSP enabled (yay) - on the ESP8266 runs just a small WifiClientToSerial program which is in the examples of the esp8266 included (it works okay for the moment until the replacement F7 arrives)
Btw to flash the chip with the IDF I had to create a new sdkconfig, which you'll find it here: https://pastebin.com/ER1S8j1C I'll hope everything there was set correctly :) 

Additional Details: FC Kakute F7 with INAV 2.1 Firmware Version Jan 8 2019 13:22:21 

So I hope you have every Info you need to help me there if you miss something just ask and I'll deliver everything possible can :) 

So Again PLEASE I need some help :D

dardrai, 2019-01-21:
If I get it to work, I'll send you all my updates to your repo (updated install instruction, photos etc and all the changes in the code - if it is necessary)

dardrai, 2019-01-21:
Update after flashing the repo manually I tried to flash the original bin files from your repo:
First flash didn't boot up because of the error: " failed: file "/home/cyber/Dokumente/Programming/esp32_SDK/esp/esp-idf/components/nvs_flash/src/nvs_item_hash_list.cpp", line 85, function: void nvs::HashList::erase(size_t) abort() was called at PC 0x401134cb on core 0"

I googled it and found out, that maybee the command could help: python esp-idf/components/esptool_py/esptool/esptool.py erase_flash
So I erased the esp and flashed it again (all values are copy paste from the readme - except the port)
python esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 --port "yourComPort" --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 bootloader.bin 0x10000 dronebridge_esp32.bin 0x8000 partitions_singleapp.bin
the ESP32 was flashed successfully and booted up just fine but the connection from the DroneBridge App is still not possible :(

seeul8er, 2019-10-16:
There is a new release that uses TCP connections and the latest SDK from Espressif. I also updated the installation guide so things should go smoothly now.

Since this issue is about the old release I'll close it. Thanks for trying out!

closed 2019-10-16 by seeul8er