DroneBridge/ESP32 #7: Not an issue: does the DroneBridge ESP32 use WifiBroadcast approach

abrilevskiy, 2020-04-09:
Does the DroneBridge ESP32 use WifiBroadcast approach (for example via esp_wifi_80211_tx)?

seeul8er, 2020-04-10:
No not as for now. It is pure Wifi with access point etc. I am thinking about adding something like that to increase range.

Maybe use the ESP-Now protocol which should be fairly similar. The upcoming ESP32-S2 will be interesting since we get USB-OTG. Allowing much higher data rates between the ESP and an end point.

abrilevskiy, 2020-04-10:
ah... I used some UART2Wifi bridge based on ESP8266 in the past based on pure Wifi - it was very inconvenient: when phone lost connection to wifi and was required to re-connect and so on...

seeul8er, 2020-04-11:
Yeah I know. But using two ESPs with a 'raw' kind of link imposes the problem that we need to get the data from the receiving ESP somehow to the phone. Bluetooth & WiFi co-existence with the ESP is a bit buggy as far as I remember.

abrilevskiy, 2020-04-11:
In order to get data from ESP to Phone and back USB-2-UART adapter could be used - have you tried such approach?
Btw, it is not in terms of the original topic, but do you aware of https://github.com/jeanleflambeur/esp32_bridge_broadcast ? Looks interesting...

closed 2020-06-18 by seeul8er