DroneBridge/ESP32 #8: Esp32-cam

adri94, 2020-06-20:
Hello,

Will this work with an esp32-cam ?
Would be nice to develop some stream and recognition.
Plus there is an ipex connector on it. 

Thank you for this nice project,

seeul8er, 2020-06-20:
Thank you,

the application should work just fine on the ESP32-Cam board.
This project does not have support for the camera. You would have to add it yourself. I am not planning on implementing it since the max. possible frame rate is too low to be usable and the only supported codec is MJPEG.

closed 2020-06-22 by seeul8er

JulesThuillier, 2022-07-18:
Hello,

What about that : https://github.com/jeanlemotan/esp32-cam-fpv ?
Framerate seems decent, and it uses WifiBroadcast too.

seeul8er, 2022-07-18:
@JulesThuillier Impressive, still too many downsides with regards to range etc.. The camera remains bad and I do not see any advantages compared to a analog setup