DroneBridge/ESP32 #10: Compatibility with ESP32 Pico D4

AaronLi, 2020-11-18:
I think I figured out an incompatibility issue while trying to run DroneBridge on my M5Stack Atom Lite.
The network would show up, but any connections to it would cause the device to restart. I think the issue is related to the ESP32 Pico's default configuration where pins 16 and 17 are used for internal storage. I resolved this by changing the default Serial pins in main.c to 21 and 25 (which make sense for my board's configuration) and then reflashing the device (after clearing the NVS).

Automatically fixing this would require being able to detect if a given ESP32 is an ESP32 Pico D4 or to just handle the error and use alternate default values

seeul8er, 2020-11-20:
Thank you for the investigation. I will not have the time to fix the issue in time. For now, I'll leave the issue open so others can see the workaround you found.

astekachev, 2021-03-04:
> I think I figured out an incompatibility issue while trying to run DroneBridge on my M5Stack Atom Lite.
> The network would show up, but any connections to it would cause the device to restart. I [...]

Hey!
Could you please show your sdkconfig-file. I have a problem with TTGO Micro-32 (ESP32 Pico D4).
Thanks!

P.S. It's bad that there is no more detailed description of how to compile the program for different boards. In principle, the description of compiling custom firmware is very bad.

seeul8er, 2021-03-04:
@astekachev There are no board-specific parameters other than the UART pins etc. so no need for any instructions. Just follow the esp-idf docs. I may need to add more info about the idf version that I used, but apart from that it is nothing special.

astekachev, 2021-03-07:
> @astekachev There are no board-specific parameters other than the UART pins etc. so no need for any instructions. Just follow the esp-idf docs. I may need to add more info about the idf version that I used, but apart from that it is nothing special.

@seeul8er I followed instructions in esp-idf docs while compiling. Unfortunately, I cannot connect to Pixracer R15 via QgraundControl or MissionPlanner use UDP or TCP. Maybe if I use a different version of ESP-IDF everything will work. Thanks!

closed 2021-07-19 by seeul8er

seeul8er, 2022-07-17:
The current release is updated with an alternative build having GPIOs 32,33 configured as default. That should fix the error.