DroneBridge/ESP32 #14: Multiple Issues

lavachemist, 2021-02-04:
I'm having multiple issues with DroneBridge on ESP32.

The first issue is that the webpage doesn't work very well. If I make changes and click save, it opens a new tab/webpage without making any changes (subsequent changes and clicks on save button result in more open tabs). It seems like I can force the changes if I click save and then power cycle the ESP32.

The next issue is that neither UDP nor TCP seems to work at all. I get an error that says connection refused when using TCP and UDP doesn't seem to do anything at all.

seeul8er, 2021-02-04:
I agree on the config web page. Back then ESP-IDF had no proper API for a HTTP server I think, so I hacked one together. The webpage should now use the HTTP Server API to make things more stable.

I know there are issues with UDP sometimes, but TCP only always worked pretty well using QGroundcontrol

lavachemist, 2021-02-04:
> I agree on the config web page. Back then ESP-IDF had no proper API for a HTTP server I think, so I hacked one together. The webpage should now use the HTTP Server API to make things more stable.
> 
> I know there are issues with UDP sometimes, but TCP alone always worked pretty well using QGroundcontrol

Interesting. I was using QGC, so maybe I'm doing something wrong. This was my second attempt though, and both attempts ended the same way.

lavachemist, 2021-02-04:
I ended up using [https://github.com/AlphaLima/ESP32-Serial-Bridge](ESP32 Serial Bridge). It works really well and is easy to set up. Posting here in case someone else has the same issues I had with ESP32 Dronebridge. Not sure why I couldn't get it to work :(

closed 2021-02-05 by seeul8er