DroneBridge/ESP32 #15: Can't connect to DroneBridge ESP32 AP. Sometimes it disappears

GeorgeFlorian, 2021-05-25:
I am having trouble connecting to the DroneBridhe ESP32 AP. I keep getting: "Couldn't get IP Address"
```
I (112188) wifi: new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (112188) wifi: station: 2e:fa:09:ac:34:43 join, AID=1, b, 20
␛[0;32mI (112268) DB_ESP32: Wifi AP started!␛[0m
I (130488) wifi: station: 2e:fa:09:ac:34:43 leave, AID = 1, bss_flags is 131121, bss:0x3ffc935c
I (130488) wifi: new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
␛[0;32mI (130488) DB_ESP32: Wifi AP stopped!␛[0m
```
I have tried two ESP32 that work with different code.

I have been trying to connect using Samsung phone and a Linux PC. 

Also, sometimes the AP doesn't appear in the WiFi list.

Some thoughts ?

seeul8er, 2021-06-03:
Hi! Could be an issue with the esp-idf. Maybe updating to a more recent version will fix the issue.

GeorgeFlorian, 2021-07-02:
> Hi! Could be an issue with the esp-idf. Maybe updating to a more recent version will fix the issue.

I have flashed it using ESP-IDF 4.2.1, which is the latest stable version as of now and I am still unable to connect to the ESP's AP.

GeorgeFlorian, 2021-07-08:
Here I am after a week of struggling with this project.

I've compiled it for my ESP32. I don't have problems with the softAP not appearing anymore.

But I have problems with connecting to it. The ESP won't assign an IP to the device that I am trying to connect with:
```
I (112188) wifi: new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (112188) wifi: station: 2e:fa:09:ac:34:43 join, AID=1, b, 20
␛[0;32mI (112268) DB_ESP32: Wifi AP started!␛[0m
I (130488) wifi: station: 2e:fa:09:ac:34:43 leave, AID = 1, bss_flags is 131121, bss:0x3ffc935c
I (130488) wifi: new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
␛[0;32mI (130488) DB_ESP32: Wifi AP stopped!␛[0m
```

softAP starts and then stops after a while.

I've tried with ESP-IDF 4.0.3, 4.2 and 4.3.

sujiv0, 2021-07-12:
Hi @GeorgeFlorian , which commit version are you using? I'm trying with v0.6, but after flashing the code no WiFi SSID is visible to connect.
Any idea?

Sujiv0

GeorgeFlorian, 2021-07-12:
> Hi @GeorgeFlorian , which commit version are you using? I'm trying with v0.6, but after flashing the code no WiFi SSID is visible to connect.
> I'm using NODEMCU ESP-32S v1.1 board.
> Any idea?
> 
> Sujiv0

I'm also using the v0.6.
Have you tried building the project for your board ?

sujiv0, 2021-07-13:
Hi @GeorgeFlorian , yeah, we tried building my code.
Now I have a doubt the in the comment for the release, it has mentioned "Configure SSID". Do I have to modify something in the code or will it take the default SSID.
After going through the code difference between v0.5 and v0.6, under "main/globals.h" file line #28 onward is something like this,
// can be set by user
extern uint8_t DEFAULT_SSID[32];
extern uint8_t DEFAULT_PWD[64];
extern uint8_t DEFAULT_CHANNEL;
extern uint8_t SERIAL_PROTOCOL;  // 1,2=MSP, 3,4,5=MAVLink/transparent

It'll be helpful if you can guide me, since I'm a newbie in ESP32 & DroneBridge.

Thanks
Sujiv0

GeorgeFlorian, 2021-07-13:
@sujiv0 I don't know how can I guide you since I couldn't make DB work and I have recently gave up on it.
I can't figure out a way of making it work.
I am sorry.

I have opened a lot of posts and threads to get some help but I still couldn't make this project work. It is beyond me.

Here are the more recent threads I've opened regarding my issues:

https://discuss.ardupilot.org/t/how-to-send-data-over-telem-1-from-cubeorange-to-esp32/73076
https://www.esp32.com/viewtopic.php?f=13&t=21788
https://www.esp32.com/viewtopic.php?f=12&t=21964
https://github.com/espressif/esp-idf/issues/7268

closed 2021-07-13 by GeorgeFlorian