DroneBridge/ESP32 #17: Can't connect GCS to DroneBridge ESP32. A lot of gibberish on Serial Monitor

GeorgeFlorian, 2021-06-23:
Hello.

I have the DroneBridge running on an ESP32 that is connected to Telem1 of CubeOrange.

I have changed the BAUD_RATE on DroneBrigde and on GCS to 115200.

I have connected to DroneBridge AP, I have opened QGroundControl, made a TCP connection to 192.168.2.1 and Port: 5760. I hit Connect and nothing happens. QGroundControl says DISCONNECTED and I get no data.

I have tried the DroneBridge app and nothing happens there.

I have the ESP connected via USB so that I can monitor it, and it sends a lot of gibberish on the Serial Monitor:
```
I (12205) wifi:station: 2e:fa:09:ac:34:43 join, AID=1, b, 20
I (12475) DB_ESP32: Wifi AP started!
�       ����    ��zd�   ��D��   �����   ��)��   ��×�    ����    ��j�   �����            ����    ���
                                                                                                        ���v�
���x�   �����   ��t��   ��G��   �����   ���1�   ��yO�   ���?�   ��A�    ��*��   �����   ��$#�   ���]�   �����   ��������    ��w��   ��I-�   ���S�    ��/��  !�����  "���B�  #��<�   $���L�  %��|2�  &��B��  '�����  (��LP�  )���.�  *�����      +��r��  ,�����  -����   .��!^�  /��� �  0���i�  1���    2��,��  3�����  4��A��  5�����  6���g�  7���    8�����      9��q��  :��O{�  ;����   <��"u�  =���
                                            �   >�����  ?����   @���Y�  A��'�   B��+��  C�����  D��F��  E�����  F���W�      G��x)�  H�����  I��v��  J��HK�  K���5�  L��%E�  M���;�  N����   O���    P��(��  Q�����  R���r�  S��
                                                                                                           �    T���|�      U��{�   V��E��  W�����  X��K`�  Y����   Z�����  [��u��  \����   ]����   ^��&n�  _����   `��@��  a�����  b����       c��~�   d����   e��q�   f��-��  g��ǌ�   h��#�   i���m�  j�����  k����   l�����  m��p��  n��N�   o���c�  p���*�      q��}T�  r��C��  s�����  t��.��  u��ħ�   v���$�  w��Z�   x�����  y����   z�� 8�  {���F�  |��M6�  }���H�  ~�����      ������  ������  ������  ���~��  ���@}�  ������  ����J�  ���s4�  ���M��  ������  ����    ��zd�   ��D��   �����       ��)��   ��×�    ����    ��j�   �����            ����    ���
                                                                        ���v�
���x�   �����   ��t��   ��G��   �����   ���1�   ��yO�   ���?�   ��A�    ��*��   �����   ��$#�   ���]�   �����   ��������    ��w��   ��I-�   ���S�    ��/��  !�����  "���B�  #��<�   $���L�  %��|2�  &��B��  '�����  (��LP�  )���.�  *�����      +��r��  ,�����  -����   .��!^�  /��� �  0���i�  1���    2��,��  3�����  4��A��  5�����  6���g�  7���    8�����      9��q��  :��O{�  ;����   <��"u�  =���
                                            �   >�����  ?����   @���Y�  A��'�   B��+��  C�����  D��F��  E�����  F���W�      G��x)�  H�����  I��v��  J��HK�  K���5�  L��%E�  M���;�  N����   O���    P��(��  Q�����  R���r�  S��
                                                                                                           �    T���|�      U��{�   V��E��  W�����  X��K`�  Y����   Z�����  [��u��  \����   ]����   ^��&n�  _����   `��@��  a�����  b����       c��~�   d����   e��q�   f��-��  g��ǌ�   h��#�   i���m�  j�����  k����   l�����  m��p��  n��N�   o���c�  p���*�      q��}T�  r��C��  s�����  t��.��  u��ħ�   v���$�  w��Z�   x�����  y����   z�� 8�  {���F�  |��M6�  }���H�  ~�����      ������  ������  ������  ���~��  ���@}�  ������  ����J�  ���s4�  ���M��  ������  ��� ��  ������  ����D�  ���:�       ������  �����   ���.X�  ����&�  ���CV�  ����(�  ������  ���}��  ����9�  ����    ���%��  ���Ϻ�   ���H��  ������      ����7�  ���vI�  ������  ���x��  ���F+�  ����U�  ���+%�  ����[�  ������  �����   ���&��  ���̑�    �����   ���l
```

How do I connect ? How can I see the telemetry data ?

astekachev, 2021-07-05:
Hi!
At first, make sure that the TELEM1 have 115200 baud rate in settings CGS, when CubeOrange connected via USB.
Then try reconnecting. However, this will not solve the problem. The connection will be unstable and more than 54% of the packets will be lost. I don't know why ESP32 works so badly. In my case, I used TTGO32 micro (Pico D4) and WROOM32. Both boards at speeds from 115200 to 921600 give the quality of reception of 46% of 100% packets MAVLink. Moreover this problem also does not depend on the UART which is used on the ESP32. That is, it's not about the hardware, but about the software. The developer doesn't really follow the project. But there is one idea. For example, ESP-01 works very well and stably with all PX4 at a speed of 921600 baud. I want to try to optimize their code for ESP32 see what happens. I do not know the ESP32 platform very well, so I will need some help.

GeorgeFlorian, 2021-07-05:
> Hi!
> At first, make sure that the TELEM1 have 115200 baud rate in settings CGS, when CubeOrange connected via USB.
> Then try reconnecting. However, this will not solve the problem. The connection will [...]

Well, at least you get something. I can't even connect QGroundControll to the ESP.
I get `I (537854) DB_CONTROL: TCP: New client connected: 192.168.2.2`
But there is nothing coming to it.

astekachev, 2021-07-05:
Try UDP connection. I used it in my case. And before that, try to swap TX and RX

GeorgeFlorian, 2021-07-05:
> Try UDP connection. I used it in my case. And before that, try to swap TX and RX

What do you mean Swap TX and RX ?
I have connected the TELEM1's TX to ESP's RX.
Is that not correct ?

seeul8er, 2021-07-05:
@GeorgeFlorian Your encoding (UTF8) or your baud-rate (between serial monitor on the PC/USB and the ESP32 serial out) do not match. That is why you see "glibberisch". You need to sort that out and then we can debug the rest. I think the ESP32 sends at 115200bps on its serial out.
What serial monitor are you using?

GeorgeFlorian, 2021-07-05:
> @GeorgeFlorian Your encoding (UTF8) or your baud-rate (between serial monitor on the PC/USB and the ESP32 serial out) do not match. That is why you see "glibberisch". You need to sort that out and [...]

Everything is at 115200.

I've build the project for my ESP32 but I can rarely connect to the softAP.
I've opened an issue her: https://github.com/espressif/esp-idf/issues/7232

Here are the current logs:
```
I (103324) DB_ESP32: Wifi AP started!
I (110754) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (110754) wifi:station: 58:96:1d:be:d4:30 join, AID=1, b, 20
I (114814) wifi:station: 58:96:1d:be:d4:30 leave, AID = 1, bss_flags is 655392, bss:0x3ffbccc4
I (114814) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (115974) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (115974) wifi:station: 58:96:1d:be:d4:30 join, AID=1, b, 20
I (162364) wifi:station: 58:96:1d:be:d4:30 leave, AID = 1, bss_flags is 655393, bss:0x3ffbccc4
I (162364) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (162364) DB_ESP32: Wifi AP started!
I (741444) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (741454) wifi:station: 2e:fa:09:ac:34:43 join, AID=1, b, 20
I (759734) wifi:station: 2e:fa:09:ac:34:43 leave, AID = 1, bss_flags is 655409, bss:0x3ffbccc4
I (759734) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (759734) DB_ESP32: Wifi AP started!
```
As you can see, my devices do not receive an IP Address.

On the rare moments that I can connect to it, I get nothing on the QGroundControll.

astekachev, 2021-07-05:
Let me try to explain my actions that I took to launch DroneBridge on my build:
1. I changed the default parameters in the main.c file
```
uint8_t DEFAULT_SSID[32] = "Pixracer";
uint8_t DEFAULT_PWD[64] = "pixracer";
uint8_t DEFAULT_CHANNEL = 11;
uint8_t SERIAL_PROTOCOL = 4;  // 1,2=MSP, 3,4,5=MAVLink/transparent
uint8_t DB_UART_PIN_TX = GPIO_NUM_1;
uint8_t DB_UART_PIN_RX = GPIO_NUM_3;
uint32_t DB_UART_BAUD_RATE = 921600;
uint16_t TRANSPARENT_BUF_SIZE = 1024;
uint8_t LTM_FRAME_NUM_BUFFER = 1;
```
2. Compiled the firmware using ESPTool version 4.0.
3. Set port TELEM1 to baud 921600
4. Then I tried to connect ESP32 to Pixracer, but due to the incorrect designation of the pinout on the ESP32 board (Сhinese fake WROOM32), I had to transfer TX to TX and RX to RX.
Ultimately, the data on the UDP began to go, but their quality was only 46%, the rest of the packages were lost.

My build:
1. Pixracer V14 (black) - TELEM1, baud set to 921600
2. TTGO Micro32 V2.0 (Pico D4) - UART0, baud 921600 (before that ESP32 WROOM DevKit)

P.S. 
In both cases, when i using two different ESP32 boards, the signal quality was poor

GeorgeFlorian, 2021-07-05:
> Let me try to explain my actions that I took to launch DroneBridge on my build:
> 
> 1. I changed the default parameters in the main.c file
> 
> ```
> uint8_t DEFAULT_SSID[32] = "Pixracer";
> uint8_t DEFAULT_PWD[64] = "pixracer";
> uint8_t DEFAULT_CHANNEL = 11;
> uint8_t SERIAL_PROTOCOL = 4;  // 1,2=MSP, 3,4,5=MAVLink/transparent
> uint8_t DB_UART_PIN_TX = GPIO_NUM_1;
> uint8_t DB_UART_PIN_RX = GPIO_NUM_3;
> uint32_t DB_UART_BAUD_RATE = 921600;
> uint16_t TRANSPARENT_BUF_SIZE = 1024;
> uint8_t LTM_FRAME_NUM_BUFFER = 1;
> ```
> 
> 1. Compiled the firmware using ESPTool version 4.0.
> 2. Set port TELEM1 to baud 921600
> 3. Then I tried to connect ESP32 to Pixracer, but due to the incorrect designation of the pinout on the ESP32 board (Сhinese fake WROOM32), I had to transfer TX to TX and RX to RX.
>    Ultimately, the data on the UDP began to go, but their quality was only 46%, the rest of the packages were lost.
> 
> My build:
> 
> 1. Pixracer V14 (black) - TELEM1, baud set to 921600
> 2. TTGO Micro32 V2.0 (Pico D4) - UART0, baud 921600 (before that ESP32 WROOM DevKit)
> 
> P.S.
> In both cases, when i using two different ESP32 boards, the signal quality was poor

Thank you for your help. but I am not even able to connect to the softAP. And even if I connect to it, the HTTP server won't work.
I am starting to give up on this project because it seems I can't make it work at all.

GeorgeFlorian, 2021-07-05:
> @GeorgeFlorian Your encoding (UTF8) or your baud-rate (between serial monitor on the PC/USB and the ESP32 serial out) do not match. That is why you see "glibberisch". You need to sort that out and [...]

I am using ESP-IDF extension on VSCode.
I have checked and `'idf.py monitor' baud rate` is set to 115200 bps.
This is in main.c `uint32_t DB_UART_BAUD_RATE = 115200;`
This is on the FlightController:
[image]

The strange thing is that not all the output is gibbeish:

```
I (1174) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
I (1284) wifi:mode : softAP (24:6f:28:de:4d:59)
I (1284) wifi:Total power save buffer number: 16
I (1284) wifi:Init max length of beacon: 752/752
I (1284) wifi:Init max length of beacon: 752/752
Am�#��j��ȭj�N[I+؟jz�M�
                     ��H֦!�MxJ�S[
�ֈH     J��
           ��HY�!>�L�
�KJ�gK�
J�wLh��
       J9LJ1�

            ���

               �ȭj*j    �

                         �H�jz�M��Z)�
                                     ��w+������ƺ�;N��
                                                   s���!���!)�
��;�XZ)�
�!)s���x�N���
ƺ�;N��
�!)s���XKt�;N��
��9{���X{i;�XZ)��Ⱦ�
                H��kʝ!��X�J��
                             @ހ�
��;���;N��

@�����혭gk      L�NNƺ�;N��J�

@����)�g�;��;I (897734) DB_CONTROL: TCP client disconnected
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect.............................
```

As you can see some of it is good. But most of it is not.

Also, I've encountered a strange behavior. While using QGroundControl on Linux and I start a TCP connection I get:
```
device reports readiness to read but returned no data (device disconnected or multiple access on port?)
Waiting for the device to reconnect.............................
```
It basically disconnects the ESP from the Serial Monitor. Why ?

Also, other errors that QGC returns are:
```
Connection to speech-dispatcher failed
VehicleLog: _waitForMavlinkMessage: collision

Vehicle 1 did not respond to request for parameters. This will cause QGroundControl to be unable to display its full user interface.
```

astekachev, 2021-07-06:
Try this in main.c

`uint16_t TRANSPARENT_BUF_SIZE = 1024;`

astekachev, 2021-07-06:
> This is in main.c `uint32_t DB_UART_BAUD_RATE = 115200;`

Try this in main.c

`uint16_t TRANSPARENT_BUF_SIZE = 1024;`

seeul8er, 2021-07-06:
OK looks like your serial monitor is good. Maybe the serial ports on the ESP32 are not configured correctly. Maybe the console & the telemetry are tying to access the same port. Maybe a newer version of the esp-idf causes problems. Did you also try the released binaries?

GeorgeFlorian, 2021-07-08:
> OK looks like your serial monitor is good. Maybe the serial ports on the ESP32 are not configured correctly. Maybe the console & the telemetry are tying to access the same port. Maybe a newer version of the esp-idf causes problems. Did you also try the released binaries?

The pre-compiled binaries won't work at all.
I've downloaded the master-branch and compiled it for my ESP32.

I don't know about the serial ports not being configured properly because when I use a simpler code that just outputs everything it gets on the serial port I get a result:
```
#define RX2 19
#define TX2 17

void setup()
{
Serial.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, RX2, TX2);
}

void loop()
{
  while(Serial2.available())
  {
    Serial.print(Serial2.read());
  }
}
``` 
Result:
```
2549521100000238133169209

2541653111110000000089253116511000178126254954110000023813312544

254955110000023813315182

2549561100000238133115205

2549571100000238133153179

254958110000023813316748

25495911000002381337778

254960110000023813320262

25496111000002381333264

254962110000023813330195

2549631100000238133244189

25416641111100000000113982249130001052282549651100000238133253108
```

So it is definitely sending information.
I've also modified RX pin to 19 inside your library, just to be sure it stays that way:
```
uint8_t DB_UART_PIN_RX = GPIO_NUM_19;
```

I don't know what else to do.

GeorgeFlorian, 2021-07-08:
I've managed to fix it somehow. I don't know what I did.

I can sometimes connect to it and sometimes I can't.

I opened another issue because I can't get parameters using MavLink.

closed 2021-07-08 by GeorgeFlorian