DroneBridge/ESP32 #19: QGC stuck at "Reading params"

GeorgeFlorian, 2021-07-08:
After weeks of trying to make this work I still can't get any parameters from the drone.

I have tried TCP and the QGC gets stuck at reading parameters.
I have tried UDP and the QGC gets stuck at reading parameters.

If and when I can connect to the softAP and establish a connection (TCP or UDP) with the ESP I get nothing from the drone.

These are the Drone settings:
[image]

I've tried both MavLink v1 and v2

Nothing works.

GeorgeFlorian, 2021-07-13:
@seeul8er Any thoughts on this ?

seeul8er, 2021-07-14:
The code is pretty old now - seems like the recent release broke some stuff. I really do not have much time at hand. I guess the HTTP Server part needs a complete rewrite to finally fix things. I think it is also a good idea to remove parts of the UDP server. That should make it simpler and more robust.
I see what I can come up with till the end of this week.

seeul8er, 2021-07-19:
Fixed in release v1.0 - #21

closed 2021-07-19 by seeul8er