DroneBridge/ESP32 #22: Feat. req.: connect to another ESP32 instead of GCS on PC/Tablet/Smartphone

rotorman, 2021-08-06:
If I understand correctly, presently DroneLink on ESP32 can connect to a PC/Tablet/Smartphone, but not to another ESP32, where two of them would emulate a wireless serial link?

If so, this is a feature request to be able to connect two or more ESP32s with DroneLink to each other to relay/route bi-directional MAVLink (transparent serial) over them.

To motivate why not connecting to a ground control station on PC/Tablet/Smartphone makes sense, please consider long range radios, mounted on a high pole for best performance. E.g. DragonLink high up on a 5m pole (see e.g. [here](https://www.rcgroups.com/forums/showthread.php?3448295-Spektrum-FrSky-Dragon-Link-Repeater) for some images of such setup). Not only need to get MAVLink data down to GCS, but also e.g. to MAVLink capable OpenTX/EdgeTX radio (e.g. http://www.olliw.eu/2020/olliwtelem/ ) and/or antenna tracker.

As latest [DragonLink](http://www.dragonlinkrc.com/instructions/v3equipment/wifisetup/) transmitter even comes with embedded user re-programmable/re-purposable ESP32, this could be super streamlined solution to install DroneLink directly there (some other pj. such as [Mavlink2Passthrough](https://github.com/zs6buj/MavlinkToPassthru/wiki) already make use of this feature), second ESP32 with DroneLink to OpenTX/EdgeTX capable radio USART. 

Connecting more than two ESP32 together, would allow even higher flexibility. Ideas for MAVLink Router can be e.g. found under: https://github.com/olliw42/fastmavlink

GCS could also join the MAVLink routing via UDP or TCP and third ESP32 with DroneLink could be for example to connect an antenna tracker.

seeul8er, 2021-08-06:
Please! It is DroneBridge not DroneLink :) The latter one sounds awefull.

> If I understand correctly, presently DroneLink on ESP32 can connect to a PC/Tablet/Smartphone, but not to another ESP32, where two of them would emulate a wireless serial link?

Yes correct. Normal Wifi. Every Wifi device can connect to the ESP32.

> If so, this is a feature request to be able to connect two or more ESP32s with DroneLink to each other to relay/route bi-directional MAVLink (transparent serial) over them.

That may work with normal Wifi - the ESP32 would need to act as an access point and station/client at the same time. Switching to the DroneBridge raw protocol (not standard WiFi anymore - more like an analog WiFi) may be more beneficial. That would increase range but would require a DroneBridge GND station or a second ESP32 running DroneBridge on the GND.

So your idea is to enable every ESP32 running DroneBridge to be configured as a relay station and/or a receiver (with UART/ output), or pure sender. That way you can 
- extend the range easily
- add new receiver devices to the system that will only require an ESP32 running DroneBridge to be able to listen to the telemetry stream
- No conversion etc. of MAVLink required
- Possible integration into DragonLink transmitter

I do see your point here. However, I don't see why you would use DB for ESP32 (Wifi) when you got DragonLink which offers way more range. If you need more range you could also switch to a more capable telemetry radio (433MHz e.g.).
But I do like the idea of a sort of mesh network that can relay data.

The thing is that I currently do not have any time at hand to implement that. I have other projects at the moment. If I start working on my drone again I will consider this idea. I already considered porting the more capable DB raw protocol to the ESP32. That will enable all these features and even more.

Best Regards,
Wolfgang

rotorman, 2021-08-06:
Hi Wolfgang,
very sorry about the misspelling - I do not know how the wrong name got so fixated in my brain...
I fixed my first post to list the correct name of your project - DroneBridge. 

Long range radio systems (LRS) are best positioned high and thus far from the typical control and feedback equipment of the pilot. Often the radio is mounted on top of a high pole, on roof of a house, caravan etc.
Another wireless link, e.g. WiFi, for the short haul makes the setup cleaner and the position of the control unit is also more free to move around if without wires.

The novel part to the equation are the native MAVLink capable radios. They have serial ports (typically TX and RX lines of an USART), but no WiFi hardware. In order to link such radio to LRS system, serial comms on both ends are required, which is why I asked if DroneBridge could connect to another DroneBridge.

When connecting more than 2 DroneBridges, they all would be bi-directional communications. A MAVLink Router would appropriately pass the telegrams between the ports. This task could be handled in code by one of the DroneBridge modules, very likely the one which is running WiFi AP where the others connect to.

Best,
Risto

seeul8er, 2021-08-07:
Ah now I get it. It's about connecting an RC to a long-range radio link somewhere (on a pole).
The ESP32 on the pole would run the current implementation of DB for ESP32 and provide an access point. The ESP in the RC would now need a transparent Wifi to UART bridge as a firmware that connects to the ESP32 on the pole and listens on the UDP/TCP port.

I think the second firmware is a rather easy implementation (even easy enough to get to know ESP/Arduino/TCP and UDP). I'd guess there might be some Arduino sketches available. Sadly I do not have time to implement the feature.

rotorman, 2021-08-07:
Hi Wolfgang,
you got the idea perfectly.

I guess the real fun and complexity would be linking more than 2 DroneBridges together - one for the LRS modem on a pole, one for the RC, one for the antenna tracker. And well, the GCS, typically on a tablet or laptop, could also attach to the communication,  but can do it directly via it's integrated WiFi module and does not need a dedicated ESP32 hw.

I understand your time constraints - whenever you would find time, I guess this would be a very interesting feature for LRS users.

I am currently working on a RC that has integrated ESP32 - basically a customized mainboard and slightly adapted EdgeTX firmware for the RadioMaster TX16S: [https://www.rcgroups.com/forums/showthread.php?3918263-Blog-22-RadioMaster-TX16S-custom-enhanced-mainboard](https://www.rcgroups.com/forums/showthread.php?3918263-Blog-22-RadioMaster-TX16S-custom-enhanced-mainboard). Running DroneBrige on it's ESP32 would be an ideal way of attaching it to MAVLink routing network.

Best,
Risto

seeul8er, 2021-10-08:
Are you sure that MAVFTP does not work? The ESP32 does a transparent link so its protocol independent,

Target0815, 2021-10-17:
Yes, I am sure ... It tries to connect via MAVFTP, but after a while the parameters are loaded individually. About 50% then comes more often "Already Got param ..." and various parameters are loaded several times and this takes ...

Current Mission Planner under Windows 10, Wifi connection to ESP32, this is connected with 115.200 Baud to SERIAL1 of the FC. Mavlink 2 is output there.

If I take a Sik-Radio pair instead of the ESP32 with Dronebridge, the parameters are loaded immediately via MAVFTP. This takes only a few seconds. The same is true for a direct USB connection, which is of course the fastest.

sebaszz, 2022-03-02:
I have more or less the same request. Could the functionality be added that dronebridge acts as a client?
In that case I can forward yaapu telemetry from my 4g/lte gcs to my rc controller with yaapu script.

seeul8er, 2022-03-02:
Sure, but that would have massive downsides on the configuration since you can only change it once the ESP32 is connected to the defined access point.
I do not have spare time at hand, nor do I personally need that feature. So if someone is interested in adding it - be my guest :)

closed 2023-10-14 by seeul8er