DroneBridge/ESP32 #23: Panic abort when using init_fs()

GeorgeFlorian, 2021-08-06:
I have compiled, built and uploaded the source code with ESP-IDF 4.4, but it doesn't work:
```
I (826) DB_ESP32: AP started!
E (826) esp_semihost: OpenOCD is not connected!
E (826) DB_ESP32: Failed to register semihost driver (ESP_ERR_NOT_SUPPORTED)!
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40088894
0x40088894: _esp_error_check_failed at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_err.c:42

file: "../main/main.c" line 260
func: app_main
expression: init_fs()

abort() was called at PC 0x40088897 on core 0
0x40088897: _esp_error_check_failed at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_err.c:43

Backtrace:0x400d344e:0x3ffbc6f00x400888a5:0x3ffbc710 0x4008edf2:0x3ffbc730 0x40088897:0x3ffbc7a0 0x400d9ffb:0x3ffbc7c0 0x40152d3f:0x3ffbc7e0 
0x400d344e: panic_abort at /home/georgeflorian/esp/esp-idf/components/esp_system/panic.c:390

0x400888a5: esp_system_abort at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_system.c:126

0x4008edf2: abort at /home/georgeflorian/esp/esp-idf/components/newlib/abort.c:46

0x40088897: _esp_error_check_failed at /home/georgeflorian/esp/esp-idf/components/esp_system/esp_err.c:43

0x400d9ffb: app_main at /home/georgeflorian/NewVolume/ESP-IDF/DroneBridgeMaster/build/../main/main.c:260 (discriminator 1)

0x40152d3f: main_task at /home/georgeflorian/esp/esp-idf/components/freertos/port/port_common.c:142
```

seeul8er, 2021-08-06:
[The developer section in the readme states that this project uses the esp-idf version 4.3](https://github.com/DroneBridge/ESP32#compile). Try compiling with that one.

seeul8er, 2021-08-06:
I think I found the issue. You will need to configure the project accordingly. Run esp-idf menuconfig and go to the DB Webserver Configuration.
Set the config to web deploy fs. The the projects SDK config here: https://github.com/DroneBridge/ESP32/blob/fcaaa58a7508d3ed4b42df203dd1606464ca184b/sdkconfig#L129

seeul8er, 2021-08-06:
I changed the default configuration to always use WEB_DEPLOY_SF. Please report if the issue is solved.

GeorgeFlorian, 2021-08-09:
> I changed the default configuration to always use WEB_DEPLOY_SF. Please report if the issue is solved.

It builds now.
Thank you.

closed 2021-08-09 by GeorgeFlorian

seeul8er, 2021-08-10:
More on the semihost topic here: https://github.com/espressif/esp-idf/blob/master/examples/protocols/http_server/restful_server/README.md#about-deploy-mode

It can make development easier since the website is not flashed to the ESP.