DroneBridge/ESP32 #24: GroundControl Softwares do not connect to ESP

GeorgeFlorian, 2021-08-06:
I have flashed the precompiled binaries on an ESP32 Wrover-B.
I have connected to the DroneBridge AP.
I have changed the UART pins accordingly.
The drone's TELECOM port is set to 115200 baud rate and Mavlink v2 (also tried v1).

I have tried connecting via UDP and TCP using MissionPlanner and QGC but it doesn't work.

When trying UDP I get `Mavlink Connecting...` loading forever. 

When trying TCP I get: `No Hearteat Packets Received`.

There is no message on the Monitor.

seeul8er, 2021-08-06:
Does the webinterface report received bytes via UART?

GeorgeFlorian, 2021-08-09:
> Does the webinterface report received bytes via UART?

Yes. There are received bytes and the number keeps growing.

I have wired the ESP better to the Drone and now I can easily connect using TCP.

UDP stays at `Mavlink Connecting...` forever and the interface stays at 0 for `Connected UDP clients`.

seeul8er, 2021-08-10:
Not sure about why UDP does not work. Did you test it with the pre-compiled version (idf 4.3)?

GeorgeFlorian, 2021-08-10:
> Not sure about why UDP does not work. Did you test it with the pre-compiled version (idf 4.3)?

Yes. It worked only once.

seeul8er, 2021-08-11:
Hard to tell from the distance why it is not working. My guess was that they might have changed something from idf 4.3 to 4.4 that causes problems now.
I'll close this for now. Let's see if others come up with similar problems. I tested it with UDP and QGC back then and it worked.

closed 2021-08-11 by seeul8er