DroneBridge/ESP32 #29: Feat. Req

ariefadha, 2022-02-06:
Can the data push to the Andruav server?
For example the esp32 connect to usb dongle modem instead becaming Access Point mode. And send the data via internet.

seeul8er, 2022-02-06:
Sure, that would be possible in theory, but it is not supported out of the box. You would need to re-write some of the code that launches the access point and instead make the ESP32 connect to your access point.
The problem is that the access point to connect to will be hardcoded inside the code. So this would be a custom solution for you and you will need to figure it out without my help :)

closed 2022-02-06 by seeul8er

ariefadha, 2022-02-06:
No need access point, esp32 act as client and connect to 3g modem (portable)

Watch this video, this guy already made it
https://youtu.be/Tg4JqZFNZJY

seeul8er, 2022-02-07:
The 3G Router does have an internal access point where the esp would connect to. However you will have to change the code of DroneBridge for ESP32 to be able to do that

jpfree, 2022-05-12:
so i  also want more network connecting，not just AP mode，like STA or STA+AP

jpfree, 2022-08-01:
> 

could you release a STA MODE version?  i am fresh for the code ,thanks