DroneBridge/ESP32 #30: DroneBridge on ground side

ryanjAA, 2022-03-04:
This may be similar to the issue posted but not clear on that and it’s a quart on rather than an issue. 

Can you plug in an esp32 (running drone bridge) to a serial modem (sik based and mavlink - take your pick) on the ground and then connect to the esp via wifi from my gcs so I can make a wifi bridge to my vehicle in the air? I Want to still use the longer range serial link but have a transparent link that is connected via wifi so I don’t have to plug into my serial modem but it is still talking to my vehicle. 

gcs (via wifi) —-w—-> esp32 (running drone bridge and hard wired to rx/tx of telem module) —> sik telemtry module ——w——-> vehicle  telemetry module -> autopilot 

is this possible?

seeul8er, 2022-03-04:
Yes, that should work. I see no issue here

ryanjAA, 2022-03-04:
What about heartbeats getting through for failsafe? That’s where I think the concern comes up.

seeul8er, 2022-03-04:
There is no parsing done on the ESP32 side. All data received is passed through. I do not see how that would affect heartbeats negatively

ryanjAA, 2022-03-05:
Ok great. I’ll give it a try and report back. Thanks!

closed 2023-10-14 by seeul8er