DroneBridge/ESP32 #31: Flashing ESP32-03-01M

NicolassSanchez, 2022-05-02:
Hello,

I am using [an ESP32-03-01M kit](https://docs.rs-online.com/81d9/A700000008496985.pdf), which is supposed to interact with a flight controller with QGroundControl.

However I cannot flash the firmware. I downloaded [the v1_1.zip](https://github.com/DroneBridge/ESP32/releases), erased the flash, and flashed the firmware with esptool folowing [these instructions](https://github.com/DroneBridge/ESP32#recommended) on Ubuntu 18.04. This gives me no error, but in the end the firmware does not appear to be flashed, as no Wi-Fi network called "DroneBridge ESP32" exists afterwards, when I restart the ESP.

I also tried with the tool on Windows 11, which tells me that the firmware has been flashed successfully has the same result.

Any idea what I am doing wrong?

Thank you!

seeul8er, 2022-05-03:
Hi!
Your hardware is not compatible with the pre-compiled binaries. They only work with ESP32 chips. You have an ESP32-C3.
See https://www.espressif.com/en/products/socs - it is very unfortunate that espressif is giving the same name prefix to all their latest products.

It should be possible to compile the firmware for the ESP32-C3 chip without an issue (maybe you will have to do some minor code/dependency changes). However I can not support you with that since I do not own such a device.

Albaref, 2022-05-03:
Ok, I see. I will try to compile it myself then, or find another solution if I don't succeed.
Thank you for your answer!

closed 2022-05-03 by Albaref