DroneBridge/ESP32 #32: TinyPICO no Wifi network visible

stefan1messmer, 2022-07-04:
I followed the instructions to flash and connect the TinyPICO V2 board, but I cannot connect to the wifi network, because I can't see it (macOS, iOS). I am new to DroneBridge and cannot figure out what's going wrong. 

Many thanks for your help
Stefan

seeul8er, 2022-07-04:
Then something went wrong during flashing. Did you flash using macOS and esptools.py? You can connect the board to your MAC and run `esptool.py -p <the_serial_port> monitor` to check the debugging output.

Consider erasing the flash memory first before flashing DroneBridge for ESP32.
`esptool.py -p <the_serial_port> erase_flash`

stefan1messmer, 2022-07-05:
`esptool.py -p <the_serial_port> monitor` gives 

>esptool: error: argument operation: invalid choice: 'monitor' (choose from 'load_ram', 'dump_mem', 'read_mem', 'write_mem', 'write_flash', 'run', 'image_info', 'make_image', 'elf2image', 'read_mac', 'chip_id', 'flash_id', 'read_flash_status', 'write_flash_status', 'read_flash', 'verify_flash', 'erase_flash', 'erase_region', 'merge_bin', 'get_security_info', 'version')

I have seen the network for a short instant, but did not manage to connect. Now it is no longer visible. Strange.

seeul8er, 2022-07-16:
Sorry for the late reply. Lots of things to do...
My bad, the esptool.py can not monitor the output. You will need to run `idf.py -p <port> monitor` or any other serial monitor that is available for your system. e.g. `miniterm.py -p PORT -b 115200`

For macOS there is e.g. https://apps.apple.com/de/app/serialtools/id611021963?mt=12
For Windows there is e.g. https://openserialportmonitor.whitestone.no/
I did not test them. The arduino serial terminal should also work

stefan1messmer, 2022-07-16:
I have installed SerialTools on macOS and connected to the port - but I cannot see anything. A test with pix32v5 connected directly to usb shows rather busy communications. I have found this:
- Defaults: UART2 (RX2, TX2 on GPIO 16, 17)

but  TinyPICO does not provide a pin with GPIO16 and GPIO17 (see below)
[image]

I even found a hint that the board will crash it one of these (16 or 17) will be addressed (cannot find it anymore). Is TinyPico really compatible with DroneBridge?

seeul8er, 2022-07-16:
You are right. Looks like TinyPico does not provide GPIO 16/17. You will have to set a different set of pins for TX/RX via the web interface. I'd try a combination of GPIO 25, 26, 27, 15, 14 or 4. You should be able to set any of them. Try TX: 25, RX 26

To debug you must connect the TinyPico to your mac via the dedicated USB (not using TX, RX pins). Debugging info will be sent via the boards USB. You do also flash TinyPico via the onboard USB.

seeul8er, 2022-07-16:
I looked up your board. It seems that GPIO 17 is connected to the flash. That would explain why it does not start up properly.
I'll see if I can come up with a fix somehow.

seeul8er, 2022-07-16:
**First, erase the flash. Otherwise, the UART pins will not be changed.**
`esptool.py -p <port> erase_flash`
Please try flashing the following firmware using the parameters 
```
0x8000 partition-table.bin
0x1000 bootloader.bin
0x10000 db_esp32.bin
0x110000 www.bin
```
[DB_ESP32_ALT_UART.zip](https://github.com/DroneBridge/ESP32/files/9126695/DB_ESP32_ALT_UART.zip)

This is a custom build that has the default UART pins changed to 
TX: GPIO33 
RX: GPIO32

stefan1messmer, 2022-07-17:
Perfect! This build works as expected. Many thanks to provide this fix for TiniPICO!

seeul8er, 2022-07-17:
Glad, that I could help

closed 2022-07-17 by seeul8er