DroneBridge/ESP32 #33: Not working with APM2.8

sir-argupta, 2022-07-16:
Hey
I tried it with apm2.8 it and its not connecting to Mission planner. Any help on this please.

seeul8er, 2022-07-16:
It is a transparent link that does not care about the protocol etc., so there must be a configuration error on your side.
You will need to provide more information if you want me to help you.
Does the WiFi show up?
Does the Web interface show data?
What's your configuration?
Is there an error message?

sir-argupta, 2022-07-16:
Does the WiFi show up? - Yes
Does the Web interface show data? - Yes
What's your configuration?

[image]

Is there an error message?

[image]

sir-argupta, 2022-07-16:
I was using with APM 2.8

seeul8er, 2022-07-16:
Very strange indeed. Could you try connecting via TCP please.
Does the baud rate of DroneBridge for ESP32 match the baud rate of your flight controller. You have set 115200 in DroneBridge, you will need to set that in the fc as well

seeul8er, 2022-07-16:
What ESP32 board are you using?

sir-argupta, 2022-07-16:
I tried both UDP & TCP but it didn't worked.. I'm using ESP WROOM32 board.

sir-argupta, 2022-07-16:
> Very strange indeed. Could you try connecting via TCP please. Does the baud rate of DroneBridge for ESP32 match the baud rate of your flight controller. You have set 115200 in DroneBridge, you will need to set that in the fc as well

Yes If you see in screenshot that I've attached..

sir-argupta, 2022-07-16:
Tho Esp8266 worked for me but I want to use esp32 as it'll provide better speed & less lagging of data.

seeul8er, 2022-07-17:
I am not familiar with AMP l, but are you sure the screenshot shows the baud that the fc has configured on its telemetry output? It is usually set via a parameter.

sir-argupta, 2022-07-17:
Yes I'm pretty sure for baud rate.

closed 2022-07-17 by sir-argupta