DroneBridge/ESP32 #34: Changing AP mode to APSTA

macnack, 2022-08-10:
hello,
I came up with idea to control several drones by QGC. My goal is to connect ESP32's to routes then in QGC switching ip to controll other one UAV.  I had change wifi mode to use Access Point and Station, but this modification prevented me from connecting over UDP. This solution works on ESP8266. Any ideas what could be blocking me?

mamariomiamo, 2022-09-01:
Hi, I was also wondering how to use **Station** mode with the ESP32. Also, in AP mode, we tried to start mavros on local machine by setting fcu_url to udp://:14550@192.168.2.1:14550 but not sure how to set the correct gcs_url as if it is set to udp://@192.168.2.2 (local machine IP), there will be a lot of messages poping up in the mavros terminal.

lv333ming, 2023-03-01:
I also encountered the problem of how to change the Station mode, no one seems to mention this problem here

jpfree, 2023-03-24:
i also hope can in sta mode....

seeul8er, 2023-10-14:
Fixed with release v1.2 - Added support for WiFi client mode (added station mode)

closed 2023-10-14 by seeul8er