DroneBridge/ESP32 #36: ESP32-C3 support

paony, 2022-11-17:
Would it be possible to compile this for ESP32-C3?
Or is the C3 hardware incompatible?
If it is possible: could you provide a binary?

seeul8er, 2022-11-17:
I think it will work just fine on the C3 module. Maybe some api differences need to be fixed, but it should work just fine.
Unfortunately I do not have the time to create the binaries myselfe.

backtozero, 2023-04-01:
> I think it will work just fine on the C3 module. Maybe some api differences need to be fixed, but it should work just fine. Unfortunately I do not have the time to create the binaries myselfe.

Trying to compile for EPS32-S3 and getting this error:
```
esp_wifi_default.h:31:41: error: unknown type name 'esp_netif_t'; did you mean 'esp_err_t'?
```

seeul8er, 2023-04-01:
looks like an API problem. You will most likely have to change the WiFi code and use the recent API calls that are supported by the S3

JustinGibDev, 2023-06-24:
I tried with a seed studio XAIO-esp32-c3 board and keeps rebooting, the GPIO pins on the board are 20,21

[image]

is there anything you can guide me in making it work with this board? I wouldnt mind re-compiling the source code if you'd point me in the right direction

seeul8er, 2023-06-25:
@JustinGibDev The WiFi initialisation is done [in this section of the code](https://github.com/DroneBridge/ESP32/blob/master/main/main.c#L139). 

[Check the esp-idf documentation on the ESP-S3](https://docs.espressif.com/projects/esp-idf/en/v5.0.2/esp32s3/api-reference/network/esp_wifi.html) with the correct esp-idf version selected on how to init WiFi.
[Check out the example implementation](https://github.com/espressif/esp-idf/blob/master/examples/wifi/getting_started/station/main/station_example_main.c) using the latest code examples for that esp-idf version.

You can hard-code the pins here if you need to. In any case, you will have to check the logs (using esp-idf toolchains "idf.py monitor" command). Otherwise, you will not be able to know what actually causes the problem.

That seedstudio board looks interesting. I might take a look at it

JustinGibDev, 2023-06-25:
Thank you very much for the advice, i will look into the code, if i get it working i'll let you know.

I have two spare esp32 board at the moment, one is the seedstudio board, and yeah i recommend you get one its quite good. Also i have a Heltec Wifi Kit 32, the crystal works at 26.7MHz and it wont run, seen on the serial monitor when it tries to boot up.  This is another option i have for now, but ideally i am going to try to get it working on the other board.

As for the seed studio : 

12:30:46.670 -> 0xffffffff
12:30:46.670 -> xffffffff
ESP-ROM:esp32c3-api1-20210207
12:30:46.670 -> Build:Feb  7 2021
12:30:46.670 -> rst:0x15 (USB_UART_CHIP_RESET),boot:0xc (SPI_FAST_FLASH_BOOT)
12:30:46.670 -> Saved PC:0x4004c0d8
12:30:46.670 -> invalid header: 0xffffffff
12:30:46.670 -> invalid header: 0xffffffff
12:30:46.670 -> invalid header: 0xffffffff

this is what i get on the serial monitor

JustinGibDev, 2023-07-08:
More of an update.  

- The setting for esp32-c3 were not preset on the sdkconfig.
- Bootloader address has to be 0x0 for the esp32-c3 on both the sdkconfig and on the esp flashtool

Currently trying to get an enviroment ready with esp-idf in order to recompile and test with  the new settings

If you would like a copy of the sdkconfig and might want to either recompile and create new binaries for the esp32-c3 specific board type

Razbotics, 2023-11-07:
Did anyone manage to recompile for esp32-c3 target?

adrian0806, 2023-11-18:
@JustinGibDev  Did you manage to get the code working on ESP32C3? I have managed to build and flash it, but the code keeps resetting

kspeco, 2023-12-14:
Hello, i recompile code with a 4.4.6 framework, flashed boot ad addr 0x0 but afeter starting ap i have a reboot.

kspeco, 2023-12-15:
Hello i have some news. I fix the reboot. Problem is in uart pins. I change pins to work with UART1 and code run but i have problem with serial socket located in db_esp32_control.c file.

int open_serial_socket() {
    int serial_socket;
    uart_config_t uart_config = {
            .baud_rate = DB_UART_BAUD_RATE,
            .data_bits = UART_DATA_8_BITS,
            .parity    = UART_PARITY_DISABLE,
            .stop_bits = UART_STOP_BITS_1,
            .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
            .rx_flow_ctrl_thresh = 0,
    };
    ESP_ERROR_CHECK(uart_param_config(UART_NUM, &uart_config));
    ESP_ERROR_CHECK(uart_set_pin(UART_NUM, 7, 6, 4, 5)); ///reboot cause
    ESP_ERROR_CHECK(uart_driver_install(UART_NUM, 1024, 0, 0, NULL, 0));
    if ((serial_socket = open("/dev/uart/2", O_RDWR)) == -1) {
        ESP_LOGE(TAG, "Cannot open UART1");
        close(serial_socket);
        uart_driver_delete(UART_NUM);
        return ESP_FAIL;
    }
    esp_vfs_dev_uart_use_driver(2);
    return serial_socket;
}
The error "Cannot open UART1" is printed. I ran out of ideas

JustinGibDev, 2023-12-17:
what is the exact board you are trying to use? probably you have to set the correct pins you might have this on the boards specific pinout sheet

seeul8er, 2023-12-27:
Release v1.3 now comes with support for the ESP32C3. Load times of the website in AP-Mode and with Desktop browsers may be an issue. Mobile android browsers performed well.

closed 2023-12-27 by seeul8er

Dhananjay2601, 2024-02-19:
> I think it will work just fine on the C3 module. Maybe some api differences need to be fixed, but it should work just fine.
> Unfortunately I do not have the time to create the binaries myselfe. 

It's not working on C3, as soon as I configure the ports to 21,20 on web interface and run a power cycle. The whole web interface crashes. I can still connect to AP but cannot access configuration from web or api.

Because of this the Esp32 C3 won't communicate with my FC (Matek H7 slim).

Help.

seeul8er, 2024-02-19:
@Dhananjay2601 Can you try the older v1.3 release? Same issue as with v1.4?

Dhananjay2601, 2024-02-19:
Yes 1.3v had same issues

seeul8er, 2024-02-19:
@Dhananjay2601 I need details on your board. my C3 works

Dhananjay2601, 2024-02-19:
Hey man you mind accepting my linkedin request? We can have a better communication on there please.

seeul8er, 2024-02-19:
@Dhananjay2601 I do not solve issues via LinkedIn nor for individuals. If you have an issue post it here so everyone can have a look at. This is no private support hotline ;)

seeul8er, 2024-02-19:
Can you post the output of: `idf.py monitor`

Dhananjay2601, 2024-02-19:
I am using Xiao ESP32-C3 by seeed studios

PS C:\Espressif\frameworks\esp-idf-v5.2> idf.py monitor
Executing action: monitor
Running cmake in directory C:\Espressif\frameworks\esp-idf-v5.2\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 C:\Espressif\frameworks\esp-idf-v5.2"...
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/mingw64/bin/gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/mingw64/bin/gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/mingw64/bin/c++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Error at CMakeLists.txt:5 (message):
  Current directory 'C:/Espressif/frameworks/esp-idf-v5.2' is not buildable.
  Change directories to one of the example projects in
  'C:/Espressif/frameworks/esp-idf-v5.2/examples' and try again.

-- Configuring incomplete, errors occurred!
See also "C:/Espressif/frameworks/esp-idf-v5.2/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\Espressif\frameworks\esp-idf-v5.2\build\log\idf_py_stderr_output_28020 and C:\Espressif\frameworks\esp-idf-v5.2\build\log\idf_py_stdout_output_28020

seeul8er, 2024-02-19:
Duplicate to #61 Will be fixed there. Seems like the D6 & D7 Pads marked for TX & RX are internally connected to UART0 which is the same UART as the one connected to the USB connector. Try any other pin combination like D9 & D10 for UART TX&RX

Dhananjay2601, 2024-02-21:
It worked! The connection to D9 and D10 solved the issue. The Matek Fc finally accepts commands from the Esp32 C3. Thanks alot for the help.