DroneBridge/ESP32 #43: idf 4.3.5 + latest firmware(default), reboot

lida2003, 2023-06-03:
I just use idf 4.3.5 + latest firmware(default), compile and flash, it reboots.

Any idea?

```
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7208
load:0x40078000,len:15132
ho 0 tail 12 room 4
load:0x40080400,len:3712
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.3.5 2nd stage bootloader
I (29) boot: compile time 15:55:34
I (29) boot: chip revision: v1.0
I (32) boot.esp32: SPI Speed      : 40MHz
I (37) boot.esp32: SPI Mode       : DIO
I (41) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (55) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (69) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (77) boot:  2 factory          factory app      00 00 00010000 00100000
I (84) boot:  3 www              Unknown data     01 82 00110000 00200000
I (92) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=1a398h (107416) map
I (143) esp_image: segment 1: paddr=0002a3c0 vaddr=3ffb0000 size=03830h ( 14384) load
I (149) esp_image: segment 2: paddr=0002dbf8 vaddr=40080000 size=02420h (  9248) load
I (153) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=8af98h (569240) map
I (364) esp_image: segment 4: paddr=000bafc0 vaddr=40082420 size=136d0h ( 79568) load
I (407) boot: Loaded app from partition at offset 0x10000
I (407) boot: Disabling RNG early entropy source...
I (419) cpu_start: Pro cpu up.
I (419) cpu_start: Starting app cpu, entry point is 0x40081334
0x40081334: call_start_cpu1 at /home/daniel/ArduPilot/esp-idf/components/esp_system/port/cpu_start.c:150

I (405) cpu_start: App cpu up.
I (435) cpu_start: Pro cpu start user code
I (435) cpu_start: cpu freq: 160000000
I (435) cpu_start: Application information:
I (440) cpu_start: Project name:     db_esp32
I (445) cpu_start: App version:      v1.0-7-g43998fd
I (451) cpu_start: Compile time:     Jun  3 2023 15:56:59
I (457) cpu_start: ELF file SHA256:  b97e7b9485607f30...
I (463) cpu_start: ESP-IDF:          v4.3.5
I (467) cpu_start: Min chip rev:     v0.0
I (472) cpu_start: Max chip rev:     v3.99
I (477) cpu_start: Chip rev:         v1.0
I (482) heap_init: Initializing. RAM available for dynamic allocation:
I (489) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (495) heap_init: At 3FFB9C90 len 00026370 (152 KiB): DRAM
I (501) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (508) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (514) heap_init: At 40095AF0 len 0000A510 (41 KiB): IRAM
I (521) spi_flash: detected chip: winbond
I (525) spi_flash: flash io: dio
W (529) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (543) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (614) DB_ESP32: Reading settings from NVS
I (634) wifi:wifi driver task: 3ffc2a7c, prio:23, stack:6656, core=0
I (634) system_api: Base MAC address is not set
I (634) system_api: read default base MAC address from EFUSE
I (644) wifi:wifi firmware version: 9ec65ef
I (644) wifi:wifi certification version: v7.0
I (644) wifi:config NVS flash: enabled
I (644) wifi:config nano formating: disabled
I (654) wifi:Init data frame dynamic rx buffer num: 32
I (654) wifi:Init management frame dynamic rx buffer num: 32
I (664) wifi:Init management short buffer num: 32
I (664) wifi:Init dynamic tx buffer num: 64
I (674) wifi:Init static rx buffer size: 1600
I (674) wifi:Init static rx buffer num: 10
I (674) wifi:Init dynamic rx buffer num: 32
I (684) wifi_init: rx ba win: 6
I (684) wifi_init: tcpip mbox: 32
I (694) wifi_init: udp mbox: 6
I (694) wifi_init: tcp mbox: 6
I (694) wifi_init: tcp tx win: 5744
I (704) wifi_init: tcp rx win: 5744
I (704) wifi_init: tcp mss: 1440
I (714) wifi_init: WiFi IRAM OP enabled
I (714) wifi_init: WiFi RX IRAM OP enabled
I (724) wifi:invalid country code string cc=XX

ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x400873dc
0x400873dc: _esp_error_check_failed at /home/daniel/ArduPilot/esp-idf/components/esp_common/src/esp_err.c:41

file: "/home/daniel/ArduPilot/DroneBridge/main/main.c" line 171
func: init_wifi
expression: esp_wifi_set_country(&wifi_country)

abort() was called at PC 0x400873df on core 0
0x400873df: _esp_error_check_failed at /home/daniel/ArduPilot/esp-idf/components/esp_common/src/esp_err.c:42

Backtrace:0x400817ce:0x3ffbc030 0x4008990d:0x3ffbc050 0x4008ff7e:0x3ffbc070 0x400873df:0x3ffbc0e0 0x400d90d4:0x3ffbc100 0x400d9607:0x3ffbc2a0 0x4015a8c3:0x3ffbc2c0
0x400817ce: panic_abort at /home/daniel/ArduPilot/esp-idf/components/esp_system/panic.c:404

0x4008990d: esp_system_abort at /home/daniel/ArduPilot/esp-idf/components/esp_system/system_api.c:112

0x4008ff7e: abort at /home/daniel/ArduPilot/esp-idf/components/newlib/abort.c:46

0x400873df: _esp_error_check_failed at /home/daniel/ArduPilot/esp-idf/components/esp_common/src/esp_err.c:42

0x400d90d4: init_wifi at /home/daniel/ArduPilot/DroneBridge/main/main.c:171 (discriminator 1)

0x400d9607: app_main at /home/daniel/ArduPilot/DroneBridge/main/main.c:261 (discriminator 2)

0x4015a8c3: main_task at /home/daniel/ArduPilot/esp-idf/components/freertos/port/port_common.c:145

ELF file SHA256: b97e7b9485607f30

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7208
load:0x40078000,len:15132
ho 0 tail 12 room 4
load:0x40080400,len:3712
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.3.5 2nd stage bootloader
I (29) boot: compile time 15:55:34

```

lida2003, 2023-06-03:
It works.

```
diff --git a/main/main.c b/main/main.c
index 5101b0e..0583fce 100644
--- a/main/main.c
+++ b/main/main.c
@@ -168,7 +168,7 @@ void init_wifi(void) {
     ESP_ERROR_CHECK(esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11B));
     ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_AP, &wifi_config));
     wifi_country_t wifi_country = {.cc = "XX", .schan = 1, .nchan = 13, .policy = WIFI_COUNTRY_POLICY_MANUAL};
-    ESP_ERROR_CHECK(esp_wifi_set_country(&wifi_country));
+    //ESP_ERROR_CHECK(esp_wifi_set_country(&wifi_country));
     ESP_ERROR_CHECK(esp_wifi_start());

     esp_netif_ip_info_t ip;

```

ThomasCr, 2023-06-04:
my ESP32 boars stucks now also in this boot loop.... I also wanted to reflash mine, but esptool stucks always on SYNC, and also `idf.py -p COM10 flash` wont connect. I connected with putty to the COM Port and there, when I press and hold the boot button and press the secound button the esp boot loop stops and displays ```
rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
```

But when I close putty and try again with esptool or odf.py I cant connect. The same when I replug the USB port and redo the key sequence... no luck...

But however, my Idea was to change the line above your commented line to my country code DE....

lida2003, 2023-06-04:
It should be problem with flashing.
The code is OK and I have got this running without problem except that line(It can be commented, no problem.)
Maybe the API(esp_wifi_set_country) should be checked or debuged. As ESP_ERROR_CHECK is very strict macro.

seeul8er, 2023-10-14:
Fixed with release v1.2 - updated country code and esp-idf version to 4.4.6

closed 2023-10-14 by seeul8er