DroneBridge/ESP32 #50: ESP32 LoRa mode

kasperg3, 2023-11-24:
Has the ESP32 Long range mode been evaluated/tested? Or else I might experiment with a long range/low data option.

The ESP32 doc states that ranges up to 1km can be reached by using their patented protocol. 

LR documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi.html#long-range-lr

seeul8er, 2023-11-25:
No, it hasn't, it requires an ESP as a receiver and a transmitter. Integrating it is very much possible, but I haven't found the time yet.

seeul8er, 2024-01-01:
Support was added with the latest commit. Function is to be tested and will be part of the new release once it works

seeul8er, 2024-05-24:
Check the latest Release Candidate for LR support: [https://github.com/DroneBridge/ESP32/releases/tag/v2.0RC1](https://github.com/DroneBridge/ESP32/releases/tag/v2.0RC1)

closed 2024-05-24 by seeul8er