DroneBridge/ESP32 #56: Wifi Client mode with no password

mcl-uk, 2024-01-29:
Seem to be unable to get dronbridge to connect (in wifi client mode) to my local, un-encrypted wifi, I tried configuring with a blank password field but dronebridge then seems unable to connect/associate. Do I have to use an excrypted wifi?

seeul8er, 2024-01-29:
Yes. The current release requires at least WPA as encryption. I may lower that in the future release.

seeul8er, 2024-01-29:
Currently there is also a bug that prevents the ESP from reconnecting on connection loss when in client mode.
I already fixed that but the code is not released yet

mcl-uk, 2024-01-29:
Thanks for the info, good to know. Will keep a look out for v1.4.

jaaptesla, 2024-02-08:
Even with password in client mode i cannot connect to 14550

I cant acces the rover unfortunately.

seeul8er, 2024-02-08:
@jaaptesla try TCP instead. TCP is very reliable. I did rewrite parts of the UDP communication in the upcoming release to make it even more stable.

jaaptesla, 2024-02-08:
> @jaaptesla try TCP instead. TCP is very reliable. I did rewrite parts of the UDP communication in the upcoming release to make it even more stable.

Thank you for response.
 i manually put in the obtained ip adress and it does connect immediately to TCP
Thank you for that.
 but im using also an APP and i cannot connect to the rover with TCP only UDP.
The app will function the same as missionplanner

If im connected to AP i just select UDP and baudrate and it connects automatically 
When i do the same when im connected to the Client IP it does not find the UDP.
So it would be Awesome if you would fix that.
Thank you for all that you do for the community your work is highly appreciated!!!

mcl-uk, 2024-02-08:
FYI, I found no problems connecting with UDP on port 14550 when connected in client mode - once I'd imposed a password on my router.

jaaptesla, 2024-02-08:
> @jaaptesla I do not think there is an easy fix for that problem. In client mode the ESP32 does **only** know the IP of the access point. It is unaware of any other devices on the network. That way it does not know where to send the UDP messages to. The issue is partly solved by the ESP32 by automatically adding all devices to the UDP-receiver list that sent data to the ESP32s UDP port. QGroundcontrol e.g does have that feature. When setting up an UDP connection you can specify a host+port in addition. QGroundcontrol will then send a packet to that hosts port. That way it 'registers' to the device (in our case the ESP32)
> 
> So in client mode you will have to do that (in your GCS) in order to be added to the list of known devices by DroneBridge. Otherwise the ESP32 will not know where to send the UDP packets to

I know that the mavlink8622 project from dogmaphic handles that wheneighter theres 20 connected to one netwerk it allows to find the udp socket.
I looked at the src but in not a coder but i tried to find the handling when connected to client wifi if it handles the udp socket.
Does it handles the udp socket opening on wifi client mode  on port 14550 ?? Can you check?

jaaptesla, 2024-02-16:
> Please check out release v1.4 - it should fix your issues

When saving the wifi client name filled in and password field as empty

It still stores a password  under: wifi_pass dronebridge
It tries to connect to the wifi client network with the password but it tries a few attempts but it does not connect to the network.
The network is online , its unencrypted it does not have a password set, yet the dronebridge is not able to connect.

I have managed to be able to store as an empty string.
But even with an empty string its able to connect once to the client network after a reboot but when i set the network offline it fails to reconnect however its says its attempting

but it seems it doesnt write the password to the NVS 
Does the partition setting allow the password to be stored as an empty string ?
Because the http rest comes up with an error saying it requires minimal 8 char.
I fixed that to zero chars but i think the nvs requires  minimal of 8.

seeul8er, 2024-02-18:
I gave this thing some thought and enabling open networks would require me to change and adapt quite a bit of code. All that effort just to enable unsecured connections. I will set the minimum to WEP and password length will remain at min. 8 chars. Since the ESP32 comes with HW-encryption support the speed penalty is also minimal.
I just to not see any benefits

closed 2024-02-18 by seeul8er

jaaptesla, 2024-02-18:
Oke i got you.