DroneBridge/ESP32 #61: Xiao ESP32-S3 and C3 targets

4O6FPV, 2024-02-18:
I have been trying to find a projects for UAS telemetry that supports the [Xiao ESP32-S3](https://www.seeedstudio.com/XIAO-ESP32S3-p-5627.html) and the [Xiao ESP32-C3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html). I have tried both the C3 and S3 precompiled targets but they both go into a constant connect and reconnect of the USB. I have tried to add them myself but I just am not able to do it with my current skill set. I was just wondering if someone could possibly make targets for these 2 board I can test. Their size and external antenna connection make these such great candidates for UAS telem. I have the board to test if needed.

seeul8er, 2024-02-18:
I placed an order for the C3, looks very interesting also for my project. I'll take a look.

seeul8er, 2024-02-19:
[Accordign to the docs](https://wiki.seeedstudio.com/XIAO_ESP32C3_Pin_Multiplexing/#special-way---use-usb-serial-and-uart0uart1-at-the-same-time), it has UART0 mapped to D6&D7, it may interfere with the USB-CDC Mode. Try configuring any other pin for UART TX&RX. E.g. use GPIO8 and GPIO9 (D8 & D9 in case of this board) for UART. 
The ESP32 can use almost any pin as a UART.

You will have to delete the flash first and re-flash the firmware to test

seeul8er, 2024-02-21:
Changing the pins seemed to do the trick #36 
Wil close this one and re-open if necessary

closed 2024-02-21 by seeul8er

4O6FPV, 2024-02-23:
so all you should have to do is change the tx/rx pin assignments after youi flash dronebridge correct? Looking at the docs on the xaio esp32-s3 it does not suffer from the same usb issue. I have gotten both to flash now just hooking to a fc to see if they speak.

Dhananjay2601, 2024-02-28:
@seeul8er I'm not able to connect to Esp32 C3 in client mode via the router using UDP 14550 in mission planner. TCP using 5760 is working fine.

Dhananjay2601, 2024-02-28:
[image]

4O6FPV, 2024-02-28:
Have you opened port 14550 within your firewall. Thus often comes up when folks have not done so to allow mp to connect in that

udp.err.png (view on web)<https://github.com/DroneBridge/ESP32/assets/111731004/7dcfc07b-8407-4f03-ba5e-8d4f77e83c9a>

seeul8er, 2024-02-28:
@Dhananjay2601 This is not an issue of DroneBridge for ESP32, but a configuration problem on your side. Mission Planner or any other app on your PC seem to already have opened up the UDP port and now you get that error message. Check your mission planner settings. Since this is not a support forum I will not further investigate