DroneBridge/ESP32 #63: Can not connect to INAV Configurator (MSP) to the FC

TByte007, 2024-02-26:
I can connect to the FC with INAV Config via the same FC cables/port with USB-TTL
I see that there is something coming on pins 17,16 of ESP32 when INAV Config is trying to connect (with the same USB-TTL and a terminal program).
I can send and receive MSP with my own ESP8266 on the same port/cables of the FC and the same speed (115200). 
I tested for continuity of the cables and it is ok/connected.

But when I try to use the ESP32 Bridge INAV Config connects then timeouts "No configuration received within 10 seconds, communication failed".
UART Serial is set to "MSP/LTM" and 11520. I don't know what "MSP & LTM to same port" means exactly but I left it default.

What could be going on ?

TByte007, 2024-02-28:
I found the problem
The line https://github.com/DroneBridge/ESP32/blob/71422b15ac54e336c97a02bb82c952d59e15cc88/main/db_esp32_control.c#L161
have to be switched with the line https://github.com/DroneBridge/ESP32/blob/71422b15ac54e336c97a02bb82c952d59e15cc88/main/db_esp32_control.c#L162
Otherwise you are always sending 0 bytes !

seeul8er, 2024-02-28:
Oh no, on the first look that seems pretty obvious to me as well. I'll fix it in the next release! Thank you!

TByte007, 2024-03-01:
> P.S: `MSP & LTM to same port` means that all messages received via the UART (no matter if LTM or MSP) are forwarded to the same TCP/UDP port. If set to no, the streams are split and sent to different ports.
Btw I don't see 1607 nor 1604 even being open UDP or TCP

> I (2485) TCP_SERVER_SETUP: Opened TCP server on port 5760
> I (2485) DB_CONTROL: Opened UDP socket on port 14550
> I (2495) DB_CONTROL: Started control module
> I (2615) DB_HTTP_REST: Starting HTTP Server
> I (2625) TCP_SERVER_SETUP: Opened TCP server on port 1603

These are the only ports being open.

seeul8er, 2024-03-01:
The data should be sent to different ports on your computer. The ESP is opening only one socket.

seeul8er, 2024-03-08:
Hopefully fixed with v1.5

closed 2024-03-08 by seeul8er