DroneBridge/ESP32 #66: The flight control board with the ESP32-C3 cannot connect to QGroundControl properly

YangChen1234, 2024-03-11:
I don't know how much to fill in the UART TX Pin and UART RX Ping on the DroneBridge page，I tried UART TX Pin = 3, UART RX Pin = 2 according to the schematic, but this did not connect to the ground station
[image]

[image]

[image]

[image]

seeul8er, 2024-03-11:
Try switching them up: TX = 2 & RX = 3.
The the web interface must show UART data being received. Otherwise QGroundControl will not be able to see anything.

YangChen1234, 2024-03-11:
I just tried to change the order and still can't connect to the ground station. How can I check the data from the serial port? thank you.
However, I connected an esp32-c3 development board outside the flight control to connect to the ground station normally
[image]
[image]

YangChen1234, 2024-03-11:
Because I want to achieve the ESP32-C3 directly welded to the flight control board, without having to attach an external development board

seeul8er, 2024-03-11:
What flight controller exactly are you using (is there more documentation)? 
Based on the schematic I cannot tell how the ESP32C3 is supposed to receive the telemetry from the flight controller chip.

YangChen1234, 2024-03-11:
[F405V3BLE-20231113.pdf](https://github.com/DroneBridge/ESP32/files/14557066/F405V3BLE-20231113.pdf)
I used AOCODARC STM32F405V3 flight control
This is the schematic, which connects BT-RX of the ESP32-C3 to TX5(PC12) of the stm32 BT-TX to RX5 (PD2) of the stm32

YangChen1234, 2024-03-11:
Could you please help me check and confirm what to fill in for UART RX Pin and UART TX Pin.Thank you very much.

YangChen1234, 2024-03-11:
[image]
[image]

seeul8er, 2024-03-11:
> [F405V3BLE-20231113.pdf](https://github.com/DroneBridge/ESP32/files/14557066/F405V3BLE-20231113.pdf) I used AOCODARC STM32F405V3 flight control This is the schematic, which connects BT-RX of the ESP32-C3 to TX5(PC12) of the stm32 BT-TX to RX5 (PD2) of the stm32

You are right according to the schematic your first configuration should be correct: RX=2 & TX=3.
If configured correctly the ESP32 should be able to receive data from the FC. This is indicated by a growing number displayed as "UART received bytes" in the top left corner of the web interface. If that number remains 0 the ESP32 is not receiving anything (and therefore is not forwarding anything to QGroundControl). This could mean the pins are not configured correctly or the FC is not sending any data because of his configuration.

So first things first: Is the "UART received bytes" indicator showing any data in the web interface?

4O6FPV, 2024-03-11:
I think there is something more going on here as my uart received bytes stops as soon as I try and connect via UDP. this is probably the same as my issue opened .

lida2003, 2024-03-12:
> I think there is something more going on here as my uart received bytes stops as soon as I try and connect via UDP. this is probably the same as my issue opened .

Thanks for the info. Is there any links about the issue? as we are going to test the link below, which we think it's passthough tunnel.
 seiral tool  ---- serial adapter ---- udp ---- udp tool

4O6FPV, 2024-03-12:
> Thanks for the info. Is there any links about the issue? as we are going to test the link below, which we think it's passthough tunnel. seiral tool ---- serial adapter ---- udp ---- udp tool

Mine was my own fault having cts/rts pins swapped which you do not even have enabled. do you have a link to the board you have looks very close to the waveshare one I have?

lida2003, 2024-03-12:
> do you have a link to the board you have looks very close to the waveshare one I have?

Do you mean our [AOCODARC F405 V3 board](https://www.aliexpress.us/item/3256806449708141.html?pdp_npi=4%40dis%21USD%21US%20%2430.45%21US%20%2430.45%21%21%21218.00%21218.00%21%40213d7a8f17102030107224943ece0f%2112000037887060766%21sh%21US%212672834312%21&spm=a2g0o.store_pc_allItems_or_groupList.new_all_items_2007616112496.1005006636022893&gatewayAdapt=glo2usa)? 

It's just on the market, and bf/inav supported, and [ardupilot](https://github.com/SnapDragonfly/ardupilot/releases/tag/Copter-4.4.4-AORC-2024.01.18) is good to go.

4O6FPV, 2024-03-12:
what esp32 are you using?

YangChen1234, 2024-03-12:
Thank you for your help. I have replaced a flight controller that can connect to the ground station.UART RX Pin =2 & UART TX Pin =3，this is fine.
[image]

[image]

closed 2024-03-12 by seeul8er