DroneBridge/ESP32 #67: Xaio ESP32-C3/waveshare esp32-s3-zero hangs when connecting via UDP

4O6FPV, 2024-03-11:
I have uploaded the V1.5 to my xiao ESP32-C3 and am able to change pins and reboot without issues now (thanks). Yet when I connect to the FC via UDP the connection hangs as shown on the webpage status the UART received bytes stopping as soon as you hit connect on UDP. I have yet to hookup my ftdi to get some debug information but here is a video showing it. I have also hooked up 7 total ESP32 variants and have only had issues with the xiao [esp32-c3](https://www.seeedstudio.com/Seeed-XIAO-ESP32C3-p-5431.html) and the [waveshare esp32-s3-zero](https://www.waveshare.com/esp32-s3-zero.htm) both stop uart received bytes when you click on connect in MP. 
https://photos.app.goo.gl/YCdV27g8aCGJFMyF7

seeul8er, 2024-03-11:
Wow that is a lot of testing, thank you. I only have a PX4 based flight controller so I only ever test with QGroundControl.
Seems like it hangs up at some point.

You be able to get sufficient logs via the USB interface. I added a chapter in the wiki regaring logging. However, hocking up an FTDI is for sure the safest way. Maybe I'll get an Ardupilot setup as well for testing

4O6FPV, 2024-03-11:
@seeul8er 
Below is from the xiao esp32-c3. I do not see anything glaring. The uart connection simply stops as soon as you hit connect via udp. I rebooted the esp32 to see if it would show anything but nothing comes across once you hit connect via udo it just stops sending uart data as the uart received bytes on webpage just stops.
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xb (SPI_FAST_FLASH_BOOT)
Saved PC:0x40388bcc
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1704
load:0x403cc710,len:0x968
load:0x403ce710,len:0x2f9c
entry 0x403cc710
I (24) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (24) boot: compile time Mar  8 2024 18:25:20
I (25) boot: chip revision: v0.3
I (27) boot.esp32c3: SPI Speed      : 80MHz
I (32) boot.esp32c3: SPI Mode       : DIO
I (37) boot.esp32c3: SPI Flash Size : 2MB
I (42) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00100000
I (80) boot:  3 www              Unknown data     01 82 00110000 00030000
I (88) boot: End of partition table
I (92) esp_image: segment 0: paddr=00010020 vaddr=3c0c0020 size=25cb8h (154808) map
I (125) esp_image: segment 1: paddr=00035ce0 vaddr=3fc91200 size=02b14h ( 11028) load
I (128) esp_image: segment 2: paddr=000387fc vaddr=40380000 size=0781ch ( 30748) load
I (137) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=b53a0h (742304) map
I (260) esp_image: segment 4: paddr=000f53c8 vaddr=4038781c size=098c8h ( 39112) load
I (273) boot: Loaded app from partition at offset 0x10000
I (274) boot: Disabling RNG early entropy source...
I (285) cpu_start: Unicore app
I (285) cpu_start: Pro cpu up.
I (294) cpu_start: Pro cpu start user code
I (294) cpu_start: cpu freq: 160000000 Hz
I (294) cpu_start: Application information:
I (297) cpu_start: Project name:     db_esp32
I (302) cpu_start: App version:      v1.4-13-g532b062-dirty
I (308) cpu_start: Compile time:     Mar  8 2024 18:25:09
I (314) cpu_start: ELF file SHA256:  0f841d7138dcb9ff...
I (320) cpu_start: ESP-IDF:          v5.1.2
I (325) cpu_start: Min chip rev:     v0.3
I (330) cpu_start: Max chip rev:     v0.99 
I (335) cpu_start: Chip rev:         v0.3
I (340) heap_init: Initializing. RAM available for dynamic allocation:
I (347) heap_init: At 3FC9A500 len 00025B00 (150 KiB): DRAM
I (353) heap_init: At 3FCC0000 len 0001C710 (113 KiB): DRAM/RETENTION
I (360) heap_init: At 3FCDC710 len 00002950 (10 KiB): DRAM/RETENTION/STACK
I (368) heap_init: At 50000010 len 00001FD8 (7 KiB): RTCRAM
I (375) spi_flash: detected chip: generic
I (379) spi_flash: flash io: dio
W (383) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (396) sleep: Configure to isolate all GPIO pins in sleep state
I (402) sleep: Enable automatic switching of GPIO sleep configuration
I (410) app_start: Starting scheduler on CPU0
I (415) main_task: Started on CPU0
I (415) main_task: Calling app_main()
I (425) DB_ESP32: Reading settings from NVS
I (425) DB_ESP32: Stored settings:
Wifi Mode: 1
ssid DroneBridge ESP32
wifi_pass dronebridge
wifi_chan 6
baud 115200u
gpio_tx 9
gpio_rx 10
gpio_cts 2
gpio_rts 3
[... 170 lines cut ...]
I (594) wifi:Init max length of beacon: 752/752
I (594) wifi:Init max length of beacon: 752/752
I (594) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (604) DB_ESP32: WIFI_EVENT - AP started! (SSID: DroneBridge ESP32 PASS: dronebridge)
I (644) DB_ESP32: MDNS Service started!
I (644) DB_ESP32: Filesystem init finished! Partition size: total: 173441 bytes, used: 61495 bytes (35%)
I (644) DB_CONTROL: Starting control module
I (654) TCP_SERVER_SETUP: Opened TCP server on port 5760
I (654) DB_CONTROL: Opened UDP socket on port 14550
I (664) DB_CONTROL: Started control module
I (1124) DB_HTTP_REST: Starting HTTP Server
I (1124) TCP_SERVER_SETUP: Opened TCP server on port 1603
I (1144) COMM: Started communication module
I (1594) main_task: Returned from app_main()
I (5314) wifi:new:<1,0>, old:<1,0>, ap:<1,0>, sta:<255,255>, prof:1
I (5314) wifi:station: ce:ae:f4:71:54:63 join, AID=1, b, 20
I (5394) DB_ESP32: WIFI_EVENT - Client connected - station:ce:ae:f4:71:54:63, AID=1
I (5694) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.2.2
I (5694) DB_ESP32: WIFI_EVENT - New station IP:192.168.2.2
I (5704) DB_ESP32: WIFI_EVENT - MAC: ce:ae:f4:71:54:63
I (19834) DB_HTTP_REST: File sending complete
I (19864) DB_HTTP_REST: File sending complete
I (29714) DB_HTTP_REST: File sending complete
I (29784) DB_HTTP_REST: File sending complete
I (29914) DB_HTTP_REST: File sending complete

seeul8er, 2024-03-11:
I see you have flow control enabled. Any difference when it is disabled? I'll check it in more detail tomorrow.

4O6FPV, 2024-03-11:
> I see you have flow control enabled. Any difference when it is disabled? I'll check it in more detail tomorrow.

to disable do you just leave rts/cts as 0 pins?

4O6FPV, 2024-03-11:
@seeul8er 

I am a idiot and had the cts/rts swapped on those to modules once I set the correct pin assignments it connected like a charm!!!!! Sorry for the screw up closing!

seeul8er, 2024-03-11:
Did the same thing during implementation, that is why I asked :)