DroneBridge/ESP32 #68: ESP32-S3-Wroom-1 hangs on initial power up

4O6FPV, 2024-03-12:
[ESP32-S3-Wroom-1](https://www.amazon.com/gp/product/B09D3S7T3M/ref=ppx_yo_dt_b_search_asin_image?ie=UTF8&psc=1) I have this board that seems to hang when you first power it. The wifi will never show up until you hit the rst button once you do that it will fire right up and broadcast the dronebridge wifi and it allows a connection with mission planner.  Below is a copy of the debug information while it is not broadcasting a wifi or anything and before I hit the RST button
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x0 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download.

Once I hit reset and then reconnect to the debug this is the following information I receive and am able to connect to the wifi and pull up the webpage. I was messing around to see if flashing from COM or USB would make a difference but it didn't. I don't see anything in the pinout and gpio pins I was using that could cause this issues. I have attached the log I hit RST to make the wifi come up. I also had reflashed this so I needed to reconfigure the pins which is also captured in the log. Maybe one of the pins I am using (gpio_tx 13, gpio_rx 14, gpio_cts 11, gpio_rts 12) is causing this issue as i don't have it on my Xiao ESP32-S3.
[esp-web-tools-logs.txt](https://github.com/DroneBridge/ESP32/files/14577986/esp-web-tools-logs.txt)
[image]

Shocktrooper, 2024-03-16:
I just ran into this myself with the same `ESP32-S3-Wroom-1`

thymjan, 2024-07-14:
Seems to be the same with the esp32c3 (connected only with RX/TX). After the first start you have to hit the RST-Button to connect to Groundstation.

seeul8er, 2024-07-14:
Please check the latest pre-release v2.0RC2 if the issue is fixed

thymjan, 2024-07-23:
I tried v2.0RC2 with esp32c3. After changing to client mode I can reach the gui. But after connecting it to Mission Planner the esp32c3 isn't reachable anymore, even after reboot. I had to write the firmware again to reach the website. But after connecting with the ground station connection is lost forever again.
So I went back to v1.5.
The connection with the ground station works again (but I have to push RST once after startup).

seeul8er, 2024-07-23:
I consider it normal that after flashing the board needs to be reset either via the RST button or by unplugging it from the power. It needs to reboot so the bootloader can select the right firmware after flashing.
With v2.0 you can press the BOOT button (on the official board it is the left button with the USB-C connector facing upwards) to reset the settings. A simple click will put the board in Access Point mode so you can change the settings. A long press will reset everything back to defaults.

@thymjan Regarding the issue with the crash: I cannot reproduce the issue with the ESP32C3. For me it works fine. I set-up the ESP32C3 in Access Point Mode. Save the settings and check that it receives telemetry etc.
Then I change it to Client Mode and enter my WiFi SSID and password. I save settings and reboot the device via the webinterface. After reboot the ESP32C3 connects to my WiFi and I am able to connect to it via the MissionPlanner using the new IP address. The web interface is also working, even when MissionPlanner is connected.
Keep in mind that the IP of the ESP32C3 changes when you switch from access point mode to client mode. How do you connect via MissionPlanner when in client mode? Via manual UDP or via TCP?

thymjan, 2024-07-24:
Ok, I will give it another try.
I have a "naked" seeed studio esp32-c3. UART TX Pin 9 and RX Pin 10. No RTS/CTS connection.
I connect Mission Planer via TCP connection.
[image]
I connect the esp32-c3 via TELEM 1 with the Radiolink crossflight V1.0 (ArduRover V4.2.3.5 ChibiOS: a8137609).
With DroneBridge version 1.5 the connection seems to be stable.
[image]

Now I will flash version "DroneBridge_ESP32_v2_0RC2\esp32c3\"
with esptool-js (first I will erase flash of the esp32c3).

thymjan, 2024-07-24:
Oh, well... 
This is working now :-)

The connection seems to be stable now.

But there are some errors:
```
2024-07-24 22:35:19 --> W (109315) httpd_txrx: httpd_sock_err: error in recv : 104
2024-07-24 22:35:58 --> I (148057) DB_CONTROL: TCP: New client connected: 192.168.1.79
2024-07-24 22:35:58 --> W (148127) DB_MAV_MSGS: GCS requested data stream with ID: 2 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148133) DB_MAV_MSGS: GCS requested data stream with ID: 2 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148137) DB_MAV_MSGS: GCS requested data stream with ID: 6 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148147) DB_MAV_MSGS: GCS requested data stream with ID: 6 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148156) DB_MAV_MSGS: GCS requested data stream with ID: 10 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148166) DB_MAV_MSGS: GCS requested data stream with ID: 10 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148178) DB_MAV_MSGS: GCS requested data stream with ID: 11 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148187) DB_MAV_MSGS: GCS requested data stream with ID: 11 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148197) DB_MAV_MSGS: GCS requested data stream with ID: 12 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148207) DB_MAV_MSGS: GCS requested data stream with ID: 12 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148217) DB_MAV_MSGS: GCS requested data stream with ID: 1 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148227) DB_MAV_MSGS: GCS requested data stream with ID: 1 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148237) DB_MAV_MSGS: GCS requested data stream with ID: 3 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:58 --> W (148540) DB_MAV_MSGS: GCS requested data stream with ID: 3 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149000) DB_MAV_MSGS: GCS requested data stream with ID: 2 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149004) DB_MAV_MSGS: GCS requested data stream with ID: 2 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149011) DB_MAV_MSGS: GCS requested data stream with ID: 6 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149021) DB_MAV_MSGS: GCS requested data stream with ID: 6 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149034) DB_MAV_MSGS: GCS requested data stream with ID: 10 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149041) DB_MAV_MSGS: GCS requested data stream with ID: 10 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149052) DB_MAV_MSGS: GCS requested data stream with ID: 11 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149061) DB_MAV_MSGS: GCS requested data stream with ID: 11 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149072) DB_MAV_MSGS: GCS requested data stream with ID: 12 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149081) DB_MAV_MSGS: GCS requested data stream with ID: 12 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149093) DB_MAV_MSGS: GCS requested data stream with ID: 1 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149101) DB_MAV_MSGS: GCS requested data stream with ID: 1 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149111) DB_MAV_MSGS: GCS requested data stream with ID: 3 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:35:59 --> W (149121) DB_MAV_MSGS: GCS requested data stream with ID: 3 and rate: 2 and start_stop: 1 - ignoring!
2024-07-24 22:36:01 --> W (151192) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 183
2024-07-24 22:36:07 --> W (157190) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:08 --> W (157715) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 183
2024-07-24 22:36:08 --> W (158160) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:08 --> W (158478) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:09 --> W (158797) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:09 --> W (159107) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:09 --> W (159424) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:10 --> W (159751) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:10 --> W (160061) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:10 --> W (160383) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:11 --> W (160701) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
[... cut ...]
2024-07-24 22:36:13 --> W (163250) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:14 --> W (163686) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:14 --> W (163904) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:14 --> W (164187) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_RESULT_MSGID_UNKNOWN msgID: 158
2024-07-24 22:36:14 --> W (164513) DB_SERIAL: fastmavlink parser had an error FASTMAVLINK_PARSE_R
```

seeul8er, 2024-07-24:
@thymjan Interesting. Did switching the TX/RX pins really resolve the issue? That is hard to believe. Thank you for the logs. At least one of the crashes can be decoded. I am not sure about the others. I'll investigate.

Regarding the parser errors: That is fine, in MAVLink mode the software now tries to register and understand MAVLink. Some GCSs like MissionPlanner however are requesting stupid stuff from us and the software keeps ignoring that. msgID 158 e.g. is either a decoding error or not defined in MAVLink. So this is not an issue. It just reduces the link quality between the GCS and the ESP a bit. It does not interfere with the other packets being sent. 
This also means that in transparent mode these errors will not appear at all since the ESP is not registering as a MAVLink device.

thymjan, 2024-07-24:
Is GPIO 9 attached to the boot button?

[https://www.esp32.com/viewtopic.php?t=27133](https://www.esp32.com/viewtopic.php?t=27133)

seeul8er, 2024-07-24:
@thymjan I was just reading about this one on another issue I try to fix. That might very well be the issue. I'll check tomorrow in more detail

thymjan, 2024-07-25:
here I found another hint using GPIO 9 for RX might be problematic:

https://wiki.seeedstudio.com/XIAO_ESP32C3_Getting_Started/#strapping-pins

seeul8er, 2024-07-25:
Thank you! I updated the docs in the Wiki!

closed 2024-08-03 by seeul8er