DroneBridge/ESP32 #70: Add custom UDP connection & UDP Broadcast

seeul8er, 2024-03-19:
- Allow users to add new UDP connections via the web interface manually. Currently, UDP implementation requires registering a UDP connection via a UDP packet in WiFi client mode.
- Users can specify a true UDP broadcast address when manually adding a new UDP connection
- Rename the current UDP implementation to directional UDP.

4O6FPV, 2024-03-19:
This would be a wonderful addition to dronebrige to work with drone swarms more specifically for light shows. 
https://skybrush.io/modules/live/

seeul8er, 2024-04-24:
A lot of effort has gone into the new release so far. I almost all of the above features already. Only thing missing is the support for static IP.
A beta version will be available soon for testing. I think DroneBridge for ESP32 will then be compatible with Skybrush. All settings are available via a REST-API so configuration of the drones can be automated for the most part

hi6otb, 2024-04-24:
@seeul8er 
Thanks.
We will test it by using our show system.

seeul8er, 2024-05-24:
RC1 of v2.0 is available: [https://github.com/DroneBridge/ESP32/releases/tag/v2.0RC1](https://github.com/DroneBridge/ESP32/releases/tag/v2.0RC1)

seeul8er, 2025-01-06:
v2.0 is out. Added UDP listening port for skybrush and static IP abilities via API

closed 2025-01-06 by seeul8er