DroneBridge/ESP32 #72: Files for official board for DroneBridge for ESP32

booo, 2024-03-28:
Hi,

thanks for the wonderful project! It really nice.

Is it possible to get the files for the offical board for DroneBride mentioned in the wiki. I would like to order some PCBs and solder the connectors to the board myself. I would appreciate it a lot if you make the design open hardware!

Best Regards
Philipp

seeul8er, 2024-03-29:
Thank you for your interest. 

I decided to not open source it or make if publicly available in the early days. For me it's also a way to compensate for some of the expenses regarding this project and also for the time it consumes developing and maintaining it.
I might hand out the files to individuals for personal use only. I am currently on holiday and have no access to my PC  until calendar week 15. If it is urgent I can check my cloud and see if I can share it that way.

booo, 2024-04-06:
Thanks for your feedback.

I would appreciate if you make the files open hardware but I can understand that you want to benefit from the project as well.

We would use the files in the context of the [SearchWIng ](https://searchwing.org) project. E.g. for cheap telemetry on test UAVs.

closed 2024-04-06 by booo