DroneBridge/ESP32 #74: API not RESTful

mmalecki, 2024-05-02:
While I was working on #73, I noticed that the API (including the one in the v2 dev branch) is not really RESTful. Would you be open to collaborating/merging a PR for making it so?

As an example:
* `GET /api/settings/request` => `GET /api/settings`
* `POST /api/settings/change` => `POST (or PATCH) /api/settings`
to achieve the typical CRUD-over-REST API style,
as well as `GET /api/system/reboot` to `POST /api/system/reboot` to satisfy the RESTful requirement of `GET` requests not having side-effects.

seeul8er, 2024-05-02:
That sounds like a good idea, if you want please go ahead and fix it in the v2_dev branch. This will becoming the new release

mmalecki, 2024-05-08:
Thanks @seeul8er. Code changes made in #76, just struggling to find out how to get the frontend built, any advice welcome :).

seeul8er, 2024-05-08:
Thanks! Unfortunately I have not setup an automated build so far. You have to manually copy & paste the index.html to the frontend folder. 
Then I usually run the `dronebridge.js` content through [minify-js](https://minify-js.com/) and paste it instead of the link to the script. I do the same for CSS content using [minifier](https://www.minifier.org/).

seeul8er, 2024-05-10:
Fixed with #77

PR #76 (2024-05-08): Make the v2.0 API RESTful
Fixes #74.
This makes changes in the `frontend_src`, but I'm struggling to find any automated process that spits out a `index.html` with the minified versions, and so wasn't able to test this on real hardware just yet. Is that build manual right now?
If so, would you be open to automating it a little? I was thinking we could minify the `.js` files (toastify and our code) separately and including them with `<script src>` tags. That leads to nicer, piece-by-piece caching (IMO, and probably wholly irrelevant here) and easier build.
closed 2024-05-10 by seeul8er