DroneBridge/ESP32 #75: Possible Discord Server?

4O6FPV, 2024-05-06:
@seeul8er,

Would you be opposed to a discord server for general chatting with no expectation that you be on it or help folks on it if you were there. As github would be the main point for any issues, enhancements or request. This would be a more general chatting for quick help with users and maybe a showcase for what you are using the product for and maybe a spot for designs others would like to share. I could create it and add whomever you wanted as admin. Just think it would be nice to have a place to chat about this project at as well as ask other users questions. Glad to see you added a donate button!!!! If you want send me a msg so we can chat about me funding some features if you are down for that.

seeul8er, 2024-05-06:
Hi,
I do support the idea of a Discord server. I did also realize that many people are not too familiar with GitHub and general questions are better answered via chat. Let's start with us two as admins and see it from there.
I will add the discord link then to the readme, wiki and the issue template so people can see.

Thank you! I didn't expect any activity on that button too soon but it already seemed to work ;) Definitely makes it easier to justify the many hours of coding that went into the project already 😄

4O6FPV, 2024-05-06:
DroneBridge or DroneBridge ESP32, for the name?

seeul8er, 2024-05-06:
Make it `DroneBridge ESP32` since the original DroneBridge project needs some overhaul and I do not want to mix them up

4O6FPV, 2024-05-06:
@seeul8er 

Here is the start. https://discord.gg/q8RJKDJY
I have gotten most permissions set just need to polish it. Join and I will set you up with admin role.

4O6FPV, 2024-05-09:
marking as closed its now operational

closed 2024-05-09 by 4O6FPV