DroneBridge/ESP32 #82: Wifi crashing

sierrayankee18, 2024-06-21:
Hello,

I have been flashing v2.0 on esp32 c3 using online flasher at correct address,  but every 20 to 25 seconds the wifi gets disconnected and again connects after a minute or so. The firmware seems to be crashing
i am trying build a drone swarm but need long range .
please provide a way out to connect 4 drones to a router for long range communication with esp32 c3 modules

thymjan, 2024-06-30:
Hi,
I observe similar problems. I'm also using a esp32 c3 with v2.0 firmware. In AP-Mode I changed the settings to Client-Mode. The esp32-c3 connects to my home network.
GUI is still usable. As soon as I connect to QGroundControl (TCP) the esp32-c3 crashes and stays in an unconnectable Mode, even after resetting.

But this is working (very stable) with an "normal" ESP32!

The actual esp32-c3 firmware seems to corrupt its memory.

sierrayankee18, 2024-06-30:
Hi Stefan.

You are right the esp32C3 fw seems to be corrupt or something else.

If you find a solution for long range mode air and ground please share.

Regards
Shravan Yadav

thymjan, 2024-06-30:
version 1.5 seems to be alright, but without the esp-now function.

sierrayankee18, 2024-06-30:
Yes I'm using that, but I'm looking to use it for long range like 1000mtr.
Let me know if any solution is available

seeul8er, 2024-07-01:
@thymjan Thank you for investigating! I will look at it.

sierrayankee18, 2024-07-01:
> @thymjan Thank you for investigating! I will look at it.

Hii,

I am gonna use it for long range telemetry.
In client mode there is a failsafe feature which gets activated when access point is unreachable for 25 seconds and switches to failsafe access point mode.
Can we add a option to turn on and off this failsafe feature in GUI, cause sometime its required and sometime its not.
I think a option will be good.

Regards

seeul8er, 2024-07-01:
@sierrayankee18 Please open a new issue to discuss that topic

seeul8er, 2024-07-14:
Please check the latest pre-release v2.0RC2 if the issue is fixed

seeul8er, 2024-07-25:
@sierrayankee18 Still crashing?

seeul8er, 2024-07-26:
Are you using GPIO9 or 8? Those seem to be problematic

closed 2024-08-03 by seeul8er

sierrayankee18, 2024-08-05:
Yes I'm using the 9 and 10 pin for data and it's crashing

seeul8er, 2024-08-06:
There was a documentation error. Do not use GPIO9. I recommend to use GPIOS 4 & 5 instead for TX and RX of the UART. The docs were updated accordingly.