DroneBridge/ESP32 #83: Falisafe Mode On/Off option for Long range telemetry

sierrayankee18, 2024-07-02:
Hii,

I am gonna use ESP32 C3 for long range telemetry.
In client mode there is a failsafe feature which gets activated when access point is unreachable for 25 seconds and switches to failsafe access point mode.
Can we add a option to turn on and off this failsafe feature in GUI, cause sometime its required and sometime its not.
I think a option will be good.

Regards

seeul8er, 2024-07-08:
I am not sure if it really needs an enhancement. You can press the Reset-Button on your ESP32 board to trigger a restart. It will then try to reconnect to the access point.
In general, I would expect the access point to be available much earlier than the ESP32. Why turn it off and on again? What is your setup/use-case? Why is the access point not available on startup?
The point of a failsafe is that it is there when you need it. Deactivating it would create a loophole.

For me that feature is rather low priority since I think it is not a problem in 90% of the cases.

sierrayankee18, 2024-07-09:
Im using it as drone telemetry, 
after 400mtr of distance from my router the esp32 connection breaks and the air borne module need to be rebooted to again connect to the router. For that i need to land the drone.(Wifi router used to increase the range)
Thats why the failsafe option was required
1. Drone esp32 C3: wifi client mode
2. Wifi router: Wifi access point

seeul8er, 2024-07-09:
I understand. We could remove the failsafe mode altogether. If someone is stuck he can soft-reset the settings by short pressing the boot button. No need for a failsafe then. Your use-case shows it's actually a big disadvantage to have it.

I can not guarantee a date. At the moment I am pushing the project to a new esp-idf version while fixing some bugs.

sierrayankee18, 2024-07-09:
I do agree with your idea of removing it completely. But there are 2 scenario.
1. Wifi access point: You can reboot it and connect again but what if you forget the passcode then you'll have to reflash the firmware and set the parameters.
2. Client mode: what if you set it to connect to a particular access point and you forget the access to wifi access point. Again you'll have to reflash the firmware.

So in my opinion for the enthusiasts who are confident in turning off the failsafe features will be useful while in development stage failsafe will be required to get it working.

So I'll recommend to have a option if enabling it and disabling.

seeul8er, 2024-07-09:
The v2.0 release implements a reset via the boot button. Short press will enable AP with default password and long press will reset the system to its defaults. So these cases are covered.

sierrayankee18, 2024-07-09:
When can we expect the release?

seeul8er, 2024-07-10:
Depends on how much free time I find and how quickly I can fix the existing bugs.
If you are lucky then this week. Maybe next week.

sierrayankee18, 2024-07-10:
i really hope if you can do it  this week

seeul8er, 2024-07-14:
Please check the latest pre-release v2.0RC2 if the issue is fixed

seeul8er, 2024-07-25:
@sierrayankee18 Fixed with v2.0RC2?

sierrayankee18, 2024-07-25:
Hii,
Just checked but its still crashing
ESP32 C3 
the wifi gets rebooted after every 5 to 7 seconds in wifi access point mode
same like the previous version

seeul8er, 2024-07-26:
I think the answer was for the other ticket. I will close this one then for now

closed 2024-07-26 by seeul8er