DroneBridge/ESP32 #85: compiling issue

pepelillo, 2024-07-11:
**Check the [Wiki](https://github.com/DroneBridge/ESP32/wiki/Troubleshooting) first!**

**Describe the bug**
compilation of master branch c3 target throws an error, seems to be and too big partition.....

"Successfully created esp32c3 image.
Generated /home/jose/Descargas/ESP32-master/build/db_esp32.bin
[941/941] cd /home/jose/Descargas/ESP32-mas...e/Descargas/ESP32-master/build/db_esp32.bin
FAILED: esp-idf/esptool_py/CMakeFiles/app_check_size /home/jose/Descargas/ESP32-master/build/esp-idf/esptool_py/CMakeFiles/app_check_size 
cd /home/jose/Descargas/ESP32-master/build/esp-idf/esptool_py && /home/jose/.espressif/python_env/idf5.2_py3.12_env/bin/python /opt/esp-idf/components/partition_table/check_sizes.py --offset 0x8000 partition --type app /home/jose/Descargas/ESP32-master/build/partition_table/partition-table.bin /home/jose/Descargas/ESP32-master/build/db_esp32.bin
Error: app partition is too small for binary db_esp32.bin size 0x100440:
  - Part 'factory' 0/0 @ 0x10000 size 0x100000 (overflow 0x440)

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**DroneBridge for ESP32 version:**
 - Release: vX.xx
 - Browser [e.g. chrome, safari]
 - ESP32 version: e.g. ESP32, ESP32C3, ESP32S3

**Additional context**
Add any other context about the problem here. Log-Output may be helpful in case of more complex issues.

seeul8er, 2024-07-11:
Hi, 
I am sorry I submitted that code to master. I am still fixing some bugs. You can fix that issue by adjusting the `partitions.csv` file. Change the line to `factory,  app,  factory, 0x10000, 1536K,`

closed 2024-07-11 by seeul8er

pepelillo, 2024-07-12:
Compilig was ok, but flashing not

esptool.py -p /dev/ttyACM0 -b 115200 --before default_reset --after hard_reset --chip esp32c3 write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 bootloader.bin 0x8000 partition-table.bin 0x10000 db_esp32.bin 0x110000 www.bin   
usage: esptool write_flash [-h] [--erase-all]
                           [--flash_freq {keep,80m,60m,48m,40m,30m,26m,24m,20m,16m,15m,12m}]
                           [--flash_mode {keep,qio,qout,dio,dout}]
                           [--flash_size {detect,keep,256KB,512KB,1MB,2MB,2MB-c1,4MB,4MB-c1,8MB,16MB,32MB,64MB,128MB}]
                           [--spi-connection SPI_CONNECTION] [--no-progress] [--verify]
                           [--encrypt]
                           [--encrypt-files <address> <filename> [<address> <filename> ...]]
                           [--ignore-flash-encryption-efuse-setting] [--force]
                           [--compress | --no-compress]
                           <address> <filename> [<address> <filename> ...]
esptool write_flash: error: argument <address> <filename>: Detected overlap at address: 0x110000 for file: www.bin

seeul8er, 2024-07-12:
after compilation flash using `idf.py flash` - it will autoselect the right parameters.
In any other case this should work as well: 
`esptool --chip esp32c3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build\bootloader\bootloader.bin 0x8000 build\partition_table\partition-table.bin 0x10000 build\db_esp32.bin 0x190000 build\www.bin`