DroneBridge/ESP32 #93: Cannot connect to Mission Planner (works on QGroundControl) in ESPNOW and in WifiLR

PeterJBurke, 2024-12-03:
Issue:

Using WifiLR in air and on ground serial USB ESP32.
Connects to QGroundcontrol but not to Mission Planner.

Using ESPnow in air and on ground serial USB ESP32:
Connects to QGroundcontrol but not to Mission Planner.

Computer:
Win 11 (Microsoft Surface Pro 9)

GCS software:
Mission Planner 1.3.82
QGroundControl 4.4.2

Flight controller and Ardupilot firmware version:
12/2/2024 4:05:29 PM : Frame: QUAD/X
12/2/2024 4:05:29 PM : RCOut: PWM:1-4 DS600:5-8 PWM:9-12
12/2/2024 4:05:29 PM : MatekF405-TE 0046003D 50315015 2033334E
12/2/2024 4:05:29 PM : ChibiOS: 6a85082c
12/2/2024 4:05:29 PM : ArduCopter V4.5.7 (2a3dc4b7)

Matek Flight Controller F405-WMN

Serial parameters:
SERIAL1_BAUD,57
SERIAL1_OPTIONS,0
SERIAL1_PROTOCOL,2

Dronebridge:

In the air:
DroneBridge for ESP32 - v2.0 RC4
ESP32C3 XIAO
Parameters exactly as shown on config page.
UART TX 5
UART RX 4
RTS PIN 0
CTS PIN 0
Protocol MAVLINK
Baud 57k
Max packet size 64
Serial read timeout (ms) 50

On the ground:
DroneBridge for ESP32 - v2.0 v2.0 RC4 SERIAL
ESP32C3 XIAO
Connected to PC on USB connector. (Not using FTDI adapter)
Parameters exactly as shown on config page.
Protocol MAVLINK
Baud 57k
Max packet size 64
Serial read timeout (ms) 50

seeul8er, 2024-12-03:
Thank you for the detailed report! This is great :D

I will have a look at it in the coming days.

seeul8er, 2024-12-08:
Hi! I was able to reproduce the error. It all works fine using the regular firmware + FTDI, but MissionPlanner failed when using the USBSerial firmware.
I will setup my debugging tools to investigate. Maybe this can be resolved on DroneBridge side. Maybe it`s something MissionPlanner has to fix.

PeterJBurke, 2024-12-09:
Some additional information:
When I am connected to qgroundcontrol, and then I quit qgroundcontrol and go over to mission planner:
I can "inspect" the mavlink packets in mission planner as follows:
Press control F. A menu comes up. Choose the mavlink inspector optoin.
You can see lots of mavlink packets coming in through dronebridge. But for some reason, mission planner does not want to connect....

rmackay9, 2024-12-09:
Hi there, I'm one of the core developers at ArduPilot and we've also noticed this issue and would like to help figure out a fix.  I've sent this over to Michael Oborne who knows best about Mission Planner to see if he has any ideas.

seeul8er, 2024-12-13:
I think I found the reason. It is a similar reason why you need to reboot the esp32 once you close QGroundControl.

When MissionPlanner opens the serial port it triggers the RTS line of the USB. This reboots the ESP32.
The solution would be to disable RTS control on the serial port in MissionPlanner so it does not reset it anymore when connecting and disconnecting.
There is already a similar option called "Connect Reset", however this setting seems to have no influence on this problem:

[image]

Alternatively (and this would be a hack, more or less), MissionPlanner could increase the timeout. The ESP32 will reboot and start sending anyway.

PeterJBurke, 2024-12-13:
Thanks for the update.
Is this something the user can do, or does it require a change in the Mission Planner code?

rmackay9, 2024-12-13:
Hi @seeul8er,

Thanks for the investigation!  I'll pass this onto the MP developers to see if they have any ideas

rmackay9, 2024-12-16:
I talked with MichaelO and he thinks that "Connect Reset" checkbox should help but he also thinks that "dts" (aka RTS/CTS) shouldn't be enabled by default.
[image]

[This is the line in MissionPlanner that uses the checkbox](https://github.com/ArduPilot/MissionPlanner/blob/master/MainV2.cs#L1560)

rmackay9, 2024-12-17:
We've also found that the checkbox has no impact on the issue so looking into alternative fixes

seeul8er, 2024-12-18:
I tested this pull request since it looked promising: https://github.com/ArduPilot/MissionPlanner/pull/3260
It allows to disable RTS control. When disabled, the ESP32C3 is no longer triggered to reboot, however, MissionPlanner still is unable to connect giving me a similar error as before. I will continue checking with MissionPlanner.

rmackay9, 2024-12-18:
Hi @seeul8er, great, thanks!  One of the MP devs is going to look into this as well once his ESP32-C3 modules arrive which should be in about a week.

rmackay9, 2024-12-29:
@seeul8er,

We've got a workaround on the MP side here now https://github.com/ArduPilot/MissionPlanner/pull/3469 and an executable can be downloaded from [this page](https://github.com/ArduPilot/MissionPlanner/actions/runs/12535485063?pr=3469).  Hopefully this change will get incorporated into MP in the near future.

I think EosBandi understands what's going on and also has some suggestions for the DroneBridge side to make it all work more smoothly.  Anyway, I think that's coming in the near-ish future.

seeul8er, 2024-12-30:
@rmackay9 Fantastic! Thank you for pushing this topic. I will make sure to update the Wiki accordingly.

closed 2025-06-18 by seeul8er