DroneBridge/ESP32 #95: [Feature Request] Add option to disable WiFi when the vehicle arms

seeul8er, 2025-01-06:
**Requested Feature:**
Disable the WiFi when the vehicle arms.

**Benefit:**
Removes the USB cable requirement to access your flight controller.

**Things to keep in mind:**
- Protocol must not be set to transparent
- ESP32 must detect arming state from the telemetry stream

Ryanf55, 2025-01-06:
This also allows you to safely use 2.4GHz for RC control. If the WiFI was turned on in flight, it would cause lots of noise. 
Thus, using the ESP32C3 on 2.4GHz would cause interference, but only when disarmed. When disarmed, you can connect wirelessly to parameters, download logs, and other usual GCS stuff on the ground.

seeul8er, 2025-01-07:
Using it as a telemetry link alongside a 2.4GHz radio link I found that interference is not a massive issue thanks to frequency hopping. It might not be ideal for latency-optimized radio protocol links, so it is nice to disable WiFi when not needed.

seeul8er, 2025-01-25:
The feature is implemented and beta-tested with MAVLink (but not MSP/LTM). It will be part of the next release.

It can be enabled via the web interface or MAVLink parameters and only works when a MAVLink or MSP/LTM are selected as serial protocol. It does not work in transparent or ESP-NOW mode (only traditional Wi-Fi modes are supported).

closed 2025-01-25 by seeul8er