DroneBridge/ESP32 #98: [Feature Request] ... GSM/LTE Support

yldzmuhammed, 2025-02-13:
**Feature Description**
Hello everyone,

First of all, I am new to this. I have cloned the repo and have been reading the code for a couple of hours but haven't seen LTE/GSM support.
What would it take to support it? Is there any document that explains the architecture for developers who want to add new features?

**Additional Context**

seeul8er, 2025-02-13:
We are currently missing the proper hardware for that. So this will not happen anytime soon.

yldzmuhammed, 2025-02-13:
Can you elaborate for me, please? What do you mean by 'missing the proper hardware', do you mean you don't have a GSM/LTE module or something else? I am open to doing the implementation but I need more documentation on the current version for that.
I need to understand the existing architecture first but I don't want to dive deep into the code for that.
Don't get me wrong, I would do it if I have to but if there are existing documents that I don't know about, that would help a lot.

yldzmuhammed, 2025-02-13:
I don't if it is my English (it's my 4th language. So, I am not great at it) or your English but I feel an attitude on your response and I really don't know what is the reason for the attitude if there is any. Ignore me if there isn't any.

I was thinking of using an LTE module like SIM7600 for it. Making LTE module functional is not that hard and I think it would be very cool and open a lot of doors.

I am not expecting anyone to do the work if it's not already there and I am fully capable to do it. But, it would be very nice if there is an arch document, it would make things more easier. But it's okay. I guess I just need to dive deep into code than.

But, I gotta say. You have done a great job by releasing this project. I am not in the drone business but this project is very nice. Well done! Congrats.

Side note: Forgive me if I misunderstood your response. Also, I appreciate the work you've done here!

seeul8er, 2025-02-14:
Yeah, it wasn't a nice response, sorry about that. 
Adding a SIM7600 will be nice, but you do not necessarily need DroneBridge with an ESP32 for that. Any microcontroller can open a UART and pass the data to the modem.
In the future, it might be worth adding a generic SPI interface specification for external radios. The ESP32 could then host a webpage for configuration.
DroneBridge (for ESP32) supporting LTE would be a completely different project overall.

closed 2025-02-21 by seeul8er