DroneBridge/ESP32 #101: ESP32 watchdog restart loop when flow control enabled

skopy123, 2025-02-22:
When I enabled flow control ESP32 keeps restarting due to watchdog resets. ESP32 boots up, connect to AP, restart within 3 seconds and cycle repeats, not able to open webUI to update configuration. this can happen when flowcontrol is enabled on droneBridge but not in FC, or flowwcontrol is configured to incorrect pin, there is no easy way out if this situation, expect command line tools and full flash erase. 

Log of from one WDT reset to another:
```
rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
I (184) boot: ESP-IDF v5.3 2nd stage bootloader
I (184) boot: compile time Jan  6 2025 16:07:11
I (184) boot: Multicore bootloader
I (189) boot: chip revision: v3.0
I (193) boot.esp32: SPI Speed      : 40MHz
I (197) boot.esp32: SPI Mode       : DIO
I (202) boot.esp32: SPI Flash Size : 2MB
W (207) boot.esp32: PRO CPU has been reset by WDT.
W (212) boot.esp32: WDT reset info: PRO CPU PC=0x40081ecf
W (218) boot.esp32: WDT reset info: APP CPU PC=0x40081ecf
I (224) boot: Enabling RNG early entropy source...
I (230) boot: Partition Table:
I (233) boot: ## Label            Usage          Type ST Offset   Length
I (241) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (248) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (256) boot:  2 factory          factory app      00 00 00010000 00180000
I (263) boot:  3 www              Unknown data     01 82 00190000 00030000
I (271) boot: End of partition table
I (275) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a004h (172036) map
I (343) esp_image: segment 1: paddr=0003a02c vaddr=3ff80000 size=00004h (     4) load
I (343) esp_image: segment 2: paddr=0003a038 vaddr=3ffb0000 size=03e68h ( 15976) load
I (355) esp_image: segment 3: paddr=0003dea8 vaddr=40080000 size=02170h (  8560) load
I (361) esp_image: segment 4: paddr=00040020 vaddr=400d0020 size=a91c4h (692676) map
I (603) esp_image: segment 5: paddr=000e91ec vaddr=40082170 size=19874h (104564) load
I (658) boot: Loaded app from partition at offset 0x10000
I (658) boot: Disabling RNG early entropy source...
I (670) cpu_start: Multicore app
I (678) cpu_start: Pro cpu start user code
I (678) cpu_start: cpu freq: 240000000 Hz
I (678) app_init: Application information:
I (681) app_init: Project name:     db_esp32
I (686) app_init: App version:      2.0 stable
I (691) app_init: Compile time:     Jan  6 2025 16:06:56
I (697) app_init: ELF file SHA256:  3b04e56caf55978f...
I (703) app_init: ESP-IDF:          v5.3
I (708) efuse_init: Min chip rev:     v0.0
I (712) efuse_init: Max chip rev:     v3.99
I (717) efuse_init: Chip rev:         v3.0
I (722) heap_init: Initializing. RAM available for dynamic allocation:
I (729) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (735) heap_init: At 3FFB9DB8 len 00026248 (152 KiB): DRAM
I (742) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (748) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (754) heap_init: At 4009B9E4 len 0000461C (17 KiB): IRAM
I (762) spi_flash: detected chip: generic
I (765) spi_flash: flash io: dio
W (769) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (783) main_task: Started on CPU0
I (787) main_task: Calling app_main()
I (816) DB_ESP32: Reading settings from NVS
I (817) DB_ESP32:       ssid: skyworks
I (818) DB_ESP32:       wifi_pass: skypass5
I (819) DB_ESP32:       ap_ip: 192.168.2.1
I (823) DB_ESP32:       ip_sta:
I (827) DB_ESP32:       ip_sta_gw:
I (830) DB_ESP32:       ip_sta_netmsk:
W (836) DB_ESP32: Could not read key udp_client_ip from NVS
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x1102 (ESP_ERR_NVS_NOT_FOUND) at 0x400db305
file: "./main/main.c" line 590
func: db_read_settings_nvs
expression: nvs_get_u16(my_handle, "udp_client_port", &udp_client_port)
I (860) DB_ESP32:       Wifi Mode: 2
        wifi_chan 6
        baud 115200u
        gpio_tx 5
        gpio_rx 4
        gpio_cts 7
        gpio_rts 6
        rts_thresh 64
        proto 4
        trans_pack_size 128
        ltm_per_packet 2
        serial_timeout 50
I (878) button: IoT Button Version: 3.4.1
I (883) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (896) wifi:wifi driver task: 3ffc206c, prio:23, stack:6656, core=0
I (910) wifi:wifi firmware version: 0caa81945
I (911) wifi:wifi certification version: v7.0
I (911) wifi:config NVS flash: enabled
I (912) wifi:config nano formating: disabled
I (916) wifi:Init data frame dynamic rx buffer num: 32
I (921) wifi:Init static rx mgmt buffer num: 5
I (925) wifi:Init management short buffer num: 32
I (930) wifi:Init dynamic tx buffer num: 64
I (934) wifi:Init static rx buffer size: 1600
I (938) wifi:Init static rx buffer num: 10
I (941) wifi:Init dynamic rx buffer num: 32
I (946) wifi_init: rx ba win: 6
I (949) wifi_init: accept mbox: 6
I (953) wifi_init: tcpip mbox: 32
I (957) wifi_init: udp mbox: 6
I (961) wifi_init: tcp mbox: 6
I (965) wifi_init: tcp tx win: 5744
I (969) wifi_init: tcp rx win: 5744
I (973) wifi_init: tcp mss: 1440
I (977) wifi_init: WiFi IRAM OP enabled
I (981) wifi_init: WiFi RX IRAM OP enabled
I (986) wifi_init: LWIP IRAM OP enabled
I (993) wifi:Set ps type: 0, coexist: 0

I (995) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
I (1063) wifi:mode : sta (94:b9:7e:55:86:70)
I (1064) wifi:enable tsf
I (1065) DB_ESP32: Init of WiFi Client-Mode finished. (SSID: skyworks PASS: *******)
I (1066) DB_ESP32: WIFI_EVENT - Wifi Started
I (1084) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (1084) wifi:state: init -> auth (0xb0)
I (1087) wifi:state: auth -> assoc (0x0)
I (1094) wifi:state: assoc -> run (0x10)
I (1131) wifi:connected with skyworks, aid = 1, channel 1, BW20, bssid = e4:8d:8c:ba:7f:59
I (1132) wifi:security: WPA2-PSK, phy: b, rssi: -48
I (1135) wifi:pm start, type: 0

I (1136) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1217) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (2146) esp_netif_handlers: sta ip: 10.0.0.97, mask: 255.255.255.0, gw: 10.0.0.1
I (2147) DB_ESP32: WIFI_EVENT - Got IP:10.0.0.97
I (2150) DB_ESP32: Connected to ap SSID:skyworks password:*******
I (2156) DB_ESP32: WiFi client mode enabled and connected!
I (2165) DB_ESP32: MDNS Service started!
I (2177) DB_ESP32: Filesystem init finished! Partition size: total: 173441 bytes, used: 80320 bytes (46%)
I (2178) DB_CONTROL: Starting control module (Wi-Fi)
I (2183) DB_SERIAL: Flow control enabled: true
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
I (184) boot: ESP-IDF v5.3 2nd stage bootloader
I (184) boot: compile time Jan  6 2025 16:07:11
I (184) boot: Multicore bootloader
```

**To Reproduce**
Steps to reproduce the behavior - minimal example:
1. Get ESP32 board, do not connect it to any flight controller, just esp32 with USB-UART bridge for programming and debug,
2. connect to AP, over web UI configure it to connect to home wifi network, save settings reset
3. open web UI over home network, set following settings  baud 115200u, gpio_tx 5,gpio_rx 4,gpio_cts 7,gpio_rts 6,rts_thresh 64,proto 4,trans_pack_size 128
4. save settings and reboot, ESP now should be in wdt restart loop.

**Expected behavior**
when this misconfiguration happend, esp should not restart, it should stay connected to AP and web UI should work to fix configuration. it would be nice if there will be warning in web UI that UART flowcontroll is not working correct or there is no data coming. 

**DroneBridge for ESP32 version:**
 - Release: v2.0 stable
 - Browser: Any
 - ESP32 version: e.g. ESP32, ESP32C6
 - AP: Mikrotik wifi b/g/n

seeul8er, 2025-02-22:
Thank you for the detailed report, this helped me a lot!

I was able to reproduce the issue with the ESP32 "classic" development board. The ESP32-C3 & C6 do not seem to have this issue.

Proposed fix: You can stop the boot loop by pressing the boot button for more than 2 seconds. This will reset all pins to defaults (0) and launch the access point.

I do agree that it would be nice to not have the ESP32 crashing and the web interface fully functional. However, I think it will be hard if not impossible. Due to the vast amount of ESP32s and their possible wirings, the config cannot be checked for plausibility before it is applied. Once the config is applied, I do not think it is possible to catch that watchdog exception. There is no flow control handshake what so ever.

Picking a working set of pins requires some care & testing when using generic or custom boards. Once a working setup was found it will not change.

seeul8er, 2025-03-29:
improved docs and exception handling. Not a bug. Cannot be fully caught

closed 2025-03-29 by seeul8er