DroneBridge/ESP32 #105: Add Indicator LED for Status & Support for Addressable LEDs

Witty-Wizard, 2025-03-17:
**Feature Description**
Currently, there is no clear visual indication of the system’s status, which makes debugging and monitoring less intuitive. Adding an indicator LED will provide immediate feedback on system states (e.g., power, connection status, errors). Additionally, support for addressable LEDs would be great

vrquaeler, 2025-06-10:
This would be a major improvement, especially for trouble shooting purposes.

seeul8er, 2025-06-10:
I do not see how a single LED could indicate all the possible problems that might be present. What would be a useful indication?

vrquaeler, 2025-06-10:
Hi Wolfgang, 
My suggestion: Make it configurable by user.
- flash or blink for "received packets" like a FTDI in Arduino
- blink for "transmitted packets" 
- blink on "MavLink hearth beat" (optional, not every body transmits MavLink)

In the past I have used a modified version of [ESP-Now-Serial-Bridge](https://github.com/yuri-rage/ESP-Now-Serial-Bridge). Here it was user's choice.

Another option is an addressable LED. Have used the "ESP-Now-Serial-Bridge" on "M5Stamp-pico" boards.

Witty-Wizard, 2025-06-11:
> Hi Wolfgang, My suggestion: Make it configurable by user.
> 
> * flash or blink for "received packets" like a FTDI in Arduino
> * blink for "transmitted packets"
> * blink on "MavLink hearth beat" (optional, not every body transmits MavLink)
> 
> In the past I have used a modified version of [ESP-Now-Serial-Bridge](https://github.com/yuri-rage/ESP-Now-Serial-Bridge). It's a transparent link only. Here the LED behavior was user's choice. Maybe it's an inspiration...
> 
> Another option is an addressable LED. Have used the "ESP-Now-Serial-Bridge" on "M5Stamp-pico" boards.

That's what exactly I had in mind with a ws2812 led driver.

skopy123, 2026-01-28:
I would also like status LED support. I may suggest two LEDs like indicators on wired network cards. One LED for connection status (like disconnected/connecting/connected) and second LED for data activity, like short blink per packet