DroneBridge/ESP32 #112: Rebooting ESP32

VasyaV7, 2025-04-29:
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/hw-reference/modules-and-boards.html#esp-modules-and-boards-esp32-wroom-32d-and-u

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080640
[0;32mI (31) boot: ESP-IDF v5.3 2nd stage bootloader[0m
[0;32mI (31) boot: compile time Jan  6 2025 16:07:11[0m
[0;32mI (31) boot: Multicore bootloader[0m
[0;32mI (35) boot: chip revision: v3.1[0m
[0;32mI (39) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (44) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (48) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (53) boot: Enabling RNG early entropy source...[0m
[0;32mI (58) boot: Partition Table:[0m
[0;32mI (62) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (84) boot:  2 factory          factory app      00 00 00010000 00180000[0m
[0;32mI (91) boot:  3 www              Unknown data     01 82 00190000 00030000[0m
[0;32mI (99) boot: End of partition table[0m
[0;32mI (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2a004h (172036) map[0m
[0;32mI (171) esp_image: segment 1: paddr=0003a02c vaddr=3ff80000 size=00004h (     4) load[0m
[0;32mI (171) esp_image: segment 2: paddr=0003a038 vaddr=3ffb0000 size=03e68h ( 15976) load[0m
[0;32mI (182) esp_image: segment 3: paddr=0003dea8 vaddr=40080000 size=02170h (  8560) load[0m
[0;32mI (188) esp_image: segment 4: paddr=00040020 vaddr=400d0020 size=a91c4h (692676) map[0m
[0;32mI (431) esp_image: segment 5: paddr=000e91ec vaddr=40082170 size=19874h (104564) load[0m
[0;32mI (485) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (486) boot: Disabling RNG early entropy source...[0m
[0;32mI (497) cpu_start: Multicore app[0m
[0;32mI (506) cpu_start: Pro cpu start user code[0m
[0;32mI (506) cpu_start: cpu freq: 240000000 Hz[0m
[0;32mI (506) app_init: Application information:[0m
[0;32mI (509) app_init: Project name:     db_esp32[0m
[0;32mI (514) app_init: App version:      2.0 stable[0m
[0;32mI (519) app_init: Compile time:     Jan  6 2025 16:06:56[0m
[0;32mI (525) app_init: ELF file SHA256:  3b04e56caf55978f...[0m
[0;32mI (531) app_init: ESP-IDF:          v5.3[0m
[0;32mI (535) efuse_init: Min chip rev:     v0.0[0m
[0;32mI (540) efuse_init: Max chip rev:     v3.99 [0m
[0;32mI (545) efuse_init: Chip rev:         v3.1[0m
[0;32mI (550) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (557) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
[0;32mI (563) heap_init: At 3FFB9DB8 len 00026248 (152 KiB): DRAM[0m
[0;32mI (569) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (576) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (582) heap_init: At 4009B9E4 len 0000461C (17 KiB): IRAM[0m
[0;32mI (590) spi_flash: detected chip: generic[0m
[0;32mI (593) spi_flash: flash io: dio[0m
[0;32mI (598) main_task: Started on CPU0[0m
[0;32mI (602) main_task: Calling app_main()[0m
[0;32mI (622) DB_ESP32: Reading settings from NVS[0m
[0;32mI (623) DB_ESP32: 	ssid: DroneBridge for ESP32[0m
[0;32mI (624) DB_ESP32: 	wifi_pass: dronebridge[0m
[0;32mI (626) DB_ESP32: 	ap_ip: 192.168.2.1[0m
[0;32mI (631) DB_ESP32: 	ip_sta: [0m
[0;32mI (634) DB_ESP32: 	ip_sta_gw: [0m
[0;32mI (638) DB_ESP32: 	ip_sta_netmsk: [0m
[0;33mW (643) DB_ESP32: Could not read key udp_client_ip from NVS[0m
ESP_ERROR_CHECK_WITHOUT_ABORT failed: esp_err_t 0x1102 (ESP_ERR_NVS_NOT_FOUND) at 0x400db305
file: "./main/main.c" line 590
func: db_read_settings_nvs
expression: nvs_get_u16(my_handle, "udp_client_port", &udp_client_port)
[0;32mI (668) DB_ESP32: 	Wifi Mode: 1
	wifi_chan 6
	baud 57600u
	gpio_tx 0
	gpio_rx 0
	gpio_cts 0
	gpio_rts 0
	rts_thresh 64
	proto 4
	trans_pack_size 128
	ltm_per_packet 2
	serial_timeout 50[0m
[0;32mI (686) button: IoT Button Version: 3.4.1[0m
[0;32mI (691) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
I (704) wifi:wifi driver task: 3ffc1f84, prio:23, stack:6656, core=0
I (712) wifi:wifi firmware version: 0caa81945
I (713) wifi:wifi certification version: v7.0
I (714) wifi:config NVS flash: enabled
I (717) wifi:config nano formating: disabled
I (722) wifi:Init data frame dynamic rx buffer num: 32
I (726) wifi:Init static rx mgmt buffer num: 5
I (730) wifi:Init management short buffer num: 32
I (735) wifi:Init dynamic tx buffer num: 64
I (739) wifi:Init static rx buffer size: 1600
I (743) wifi:Init static rx buffer num: 10
I (747) wifi:Init dynamic rx buffer num: 32
[0;32mI (752) wifi_init: rx ba win: 6[0m
[0;32mI (754) wifi_init: accept mbox: 6[0m
[0;32mI (758) wifi_init: tcpip mbox: 32[0m
[0;32mI (762) wifi_init: udp mbox: 6[0m
[0;32mI (766) wifi_init: tcp mbox: 6[0m
[0;32mI (770) wifi_init: tcp tx win: 5744[0m
[0;32mI (774) wifi_init: tcp rx win: 5744[0m
[0;32mI (778) wifi_init: tcp mss: 1440[0m
[0;32mI (782) wifi_init: WiFi IRAM OP enabled[0m
[0;32mI (787) wifi_init: WiFi RX IRAM OP enabled[0m
[0;32mI (791) wifi_init: LWIP IRAM OP enabled[0m
I (802) wifi:set country: cc=US

VasyaV7, 2025-04-29:
I (804) wifi:Set ps type: 0, coexist: 0

[0;32mI (806) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08[0m
I (876) wifi:mode : softAP (1c:69:20:c7:0a:b9)
I (880) wifi:Total power save buffer number: 32
I (881) wifi:Init max length of beacon: 752/752
I (882) wifi:Init max length of beacon: 752/752
[0;32mI (885) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1[0m
[0;32mI (892) DB_ESP32: WIFI_EVENT - AP started! (SSID: DroneBridge for ESP32 PASS: dronebridge)[0m
[0;32mI (905) DB_ESP32: MDNS Service started![0m
[0;31mE (906) SPIFFS: mount failed, -10025[0m
[0;31mE (909) DB_ESP32: Failed to mount or format filesystem[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x400db50a
file: "./main/main.c" line 751
func: app_main
expression: init_fs()

abort() was called at PC 0x4008a42f on core 0

Backtrace: 0x40081e6e:0x3ffbbfd0 0x4008a439:0x3ffbbff0 0x4009480a:0x3ffbc010 0x4008a42f:0x3ffbc080 0x400db50a:0x3ffbc0b0 0x40172703:0x3ffbc0d0

ELF file SHA256: 3b04e56caf55978f

Rebooting...
ets Jul 29 2019 12:21:46

VasyaV7, 2025-04-29:
https://docs.espressif.com/projects/esp-idf/en/release-v4.2/esp32/hw-reference/modules-and-boards.html#esp-modules-and-boards-esp32-wroom-32d-and-u

seeul8er, 2025-05-01:
Did you flash via the online flasher?

VasyaV7, 2025-05-01:
No, when flashing via the online flasher, the board keeps rebooting. These errors appear when flashing using esptool.

seeul8er, 2025-05-01:
Make sure you flash the partition table correctly. The online flasher should do that automatically, causing no issues. You have a custom board?

VasyaV7, 2025-05-01:
I am using an ESP32: https://sl.aliexpress.ru/p?key=yMj33HH

seeul8er, 2025-05-01:
I have the legacy ESP32 as well, and it works.
Due to export control regulations, I cannot give you any further support.

closed 2025-05-01 by seeul8er